USBGuard/usbguard #22: [RFE] allow a usb device with a mass storage interface but ensure it'll be mounted read-only

dkopecek, 2015-04-14:
This could be used to prevent data leakage without the strict restriction of "no usb mass storage devices can be connected to the system". Obviously, this would be effective only for user without root access.

dkopecek, 2015-04-15:
It is possible to set read-only flag for block devices using blockdev --setro /dev/sdX. Internally, it calls:

```
open("/dev/sdX", O_RDONLY); => 3
ioctl(3, BLKROSET, 1);
close(3);
```

So it's quite trivial to implement.

dkopecek, 2015-04-15:
This is probably a good candidate for a new keyword in the action section in a rule. E.g.:
```
allow 1234:1234 serial "abcdef" with-interface { 08:*:* } set-blockdev-ro
```
The `set-blockdev-ro` simply means that any block devices created as a result of inserting a device that matches the rule will be set to read-only mode.

dkopecek, 2015-04-15:
The use case for this is data leakage prevention, i.e. if you want to allow access to USB flash disks but you don't wan't unprivileged users to be able to copy data onto the device from the system. This idea was originally proposed to me by jpokorny@r-----.c--.

dkopecek, 2015-04-15:
Alternative names for the keyword:
 * set-read-only
 * force-read-only

the `set-blockdev-ro` name has the advantage of implying that the action is applicable to block devices only...

jnpkrn, 2015-04-15:
Great to see it's feasible :)

flavienbwk, 2020-09-10:
Is it still an issue or has it been introduced ?

Idriel, 2021-08-06:
is it possible to vote for this. I don't see option for this feature and it would be nice to have it because of compliance requests also?