USBGuard/usbguard #26: [RFE] Create a SELinux policy for usbguard

dkopecek, 2015-04-14:
The code isn't bullet proof. Let's try to secure the daemon with all that's available.

dkopecek, 2015-08-25:
Related ticket in the pgl project: https://github.com/dkopecek/pgl/issues/9

vinzent, 2017-08-31:
@dkopecek  i've created a initial policy for Fedora - https://github.com/fedora-selinux/selinux-policy-contrib/pull/26

problems i can see:

* the daemon won't start if he can't open usbguard-daemon.conf read/write. For servers I don't want to allow usbguard to manage usbguard-daemon.conf. 
* desktop vs. server: managing the config with usbguard-applet-qt or usbguard is desirable on desktops but IMHO not on servers. what should be the default of the selinux policy? allow write access to the usbguard-daemon.conf and the rules.conf

BenjaminSchubert, 2017-08-31:
Desktop vs. server : I would say it is better to have it open so that desktop users won't have any problems, as server administrators would be more at ease tweaking their selinux policy if they need it, but that's just my opinion.

Otherwise, would it be possible to have the usbguard-applet-qt installing a patch to the selinux rules to allow access ? Therefore usbguard-daemon alone => no write access, with applet => read-write access ? Don't know if this is possible nor wanted though :)

vinzent, 2017-09-01:
@benjaminschubert I agree with you that the default should be to allow the write by usbguard daemon. Because otherwise i think people will rather blame selinux and disable it completly.

dkopecek, 2017-09-01:
@vinzent Hello and thanks for this contribution! I've changed the ConfigFile class so that is can open the file in read-only mode, see b0ef90c1acb8f7befad15ac349d2011217e6226f and 2f60391fc03b204380e9e87c4c1022fa48198ac2.

In case of the rule file, I would prefer to have a selinux boolean that can be set/unset to choose between the two scenarios. Distros can then ship with the correct setting based on what product of the distro is being installed (Workstation, Server, ...). What do you think @BenjaminSchubert @vinzent @wrabcak @bachradsusi?

wrabcak, 2018-09-06:
I took sources from @vinzent  PR  (https://github.com/fedora-selinux/selinux-policy-contrib/pull/26) and created PR in usbgurad-selinux repo (https://github.com/USBGuard/usbguard-selinux/pull/1). 

It would be great to create rpm subpackage (e.g: usbgurad-selinux) based on this packaging template:
https://fedoraproject.org/wiki/PackagingDrafts/SELinux_Independent_Policy#sec-removing_product_policy

If you have any questions/troubles feel free to contact me.

BirdInFire, 2022-02-02:
> I took sources from @vinzent PR ([fedora-selinux/selinux-policy-contrib#26](https://github.com/fedora-selinux/selinux-policy-contrib/pull/26)) and created PR in usbgurad-selinux repo ([USBGuard/usbguard-selinux#1](https://github.com/USBGuard/usbguard-selinux/pull/1)).
> 
> It would be great to create rpm subpackage (e.g: usbgurad-selinux) based on this packaging [template](https://fedoraproject.org/wiki/PackagingDrafts/SELinux_Independent_Policy#sec-removing_product_policy).
> 
> If you have any questions/troubles feel free to contact me.

i have a question, on actual Fedora 35 does usbguard-selinux is complete or does it have some change to be done (to harden-it or fixing bug) (i ask to see if i just install and forget) or if i search to PR any bug i see.

Thanks

radosroka, 2022-02-03:
> i have a question, on actual Fedora 35 does usbguard-selinux is complete or does it have some change to be done (to harden-it or fixing bug) (i ask to see if i just install and forget) or if i [...]

The usbguard-selinux is present on f35. This ticket should be closed.

closed 2022-02-03 by radosroka