USBGuard/usbguard #27: [RFE] Allow to set a rate limit for device/rule processing

dkopecek, 2015-04-15:
To slow down bruteforce attacks on attribute values (e.g. iSerial value)

dkopecek, 2016-03-25:
One option to implement rate limiting is to use rule-applied/rule-evaluated conditions, e.g.:
```
allow 1234:567 if !rule-evaluated(10)
```
This rule will apply only if it wasn't evaluated in the last 10 seconds.

A "global" rate limit could be implemented using a conditioned block rule as the first rule in the policy, e.g.:
```
block if rule-evaluated(5)
...other rules...
```
However, a device which will hit such a rule will not be re-evaluated again after the required time will pass (5 seconds in this case).

That could be problematic if one would like to connect a hub with several devices attached to it. In that case, the block rule would not apply to the hub device but it would be applied to the devices attached to it because they'll get processed immediately after the hub device and therefore they won't meet the 5 second delay condition.

radosroka, 2019-08-08:
We should probably close this issue @dkopecek.

dkopecek, 2019-08-10:
Yes!

closed 2019-08-10 by dkopecek