USBGuard/usbguard #34: [RFE] When explicitly asked for using a rule, monitor the traffic to/from a USB HID device

dkopecek, 2015-05-29:
Since it is usually hard to differentiate between a real keyboard and a faked one (no iSerial value exported, mobile users tend to use several keyboard and leave them at the places they move from/to, ...) we could at least look for clues in the USB traffic, i.e. behaviour of the device. The Linux kernel provides a facility to monitor USB traffic, [usbmon](https://www.kernel.org/doc/Documentation/usb/usbmon.txt).

Rogue keyboard devices usually try to perform their task fast (with some initial delay, or when triggered by an external event -- lighting conditions, RF remote control, ...) so as not to be noticed by the user of the targeted computer. And they use a limited set of characters (scan codes).

A normal keyboard on the other hand, when used by a human, sends all kinds of characters -- the statistical properties of such traffic probably highly depend on the operator.

USBGuard could monitor the traffic of selected devices and notify the user when something unusual happens or when it classifies the traffic as originating from a rouge device. Some learning step might be necessary  to prevent a lot of false positives...

dkopecek, 2015-07-29:
Related:
http://slashdot.org/story/15/07/28/1440210/chrome-extension-thwarts-user-profiling-based-on-typing-behavior

dkopecek, 2015-07-29:
One of the use cases is also monitoring of outgoing traffic to a USB flash stick (or whatever) for auditing purposes.

Unusually large traffic to a device could indicate an event of unauthorized data exfiltration. We could probably also implement the capability of limiting i/o traffic to certain devices in the rule language, e.g. when a device exceeds the limit, it'll be blocked, disconnected, blacklisted.

dkopecek, 2016-01-17:
Related:
https://github.com/radosroka/usbmon

indolering, 2017-02-18:
This sounds a lot like anti-virus heuristic scanning.  Sure, you can theoretically catch naive implementations but it's a lot of work for something that's trivial to circumvent and likely to lead to a lot of false positives.  I just don't see this as a high priority item....