USBGuard/usbguard #38: [RFE] Integrate usbguard into Freeipa

rexcze, 2015-06-15:
Freeipa (www.freeipa.org) is centralized management tool for maintaining identity and access policy like HBAC or SUDO with help of sssd on client machines. I think that managing sudo policies is very similar to usbguard policies and thus having IPA user interface for usbguard could simplify management on large deployments (government, enterprises) and could extend freeipa's functionality. I think that this RFE is blocked by issue no. 30, because we need to be able to allow usb devices based on users and computers.

For example user1 is allowed to use USB Flash on computer1, but is not allowed to use it on computer2. 
Another example could be, there is some USB device which is allowed on all computers. 

Integration with Freeipa could be managed by creating Freeipa module. https://www.freeipa.org/images/5/5b/FreeIPA33-extending-freeipa.pdf

dkopecek, 2016-04-22:
Just for the record, this is related to issue #43.

dkopecek, 2016-04-26:
Filled FreeIPA upstram ticket: https://fedorahosted.org/freeipa/ticket/5850

closed 2019-03-11 by dkopecek