USBGuard/usbguard #44: Parse error - 1 is not a number

dx486, 2015-11-02:
I have set up usbguard following the guide [here](https://github.com/dkopecek/usbguard/blob/master/doc/usbguard-on-ubuntu-14.04.md).

When I run usbguard-applet-qt I get this message on terminal: 

> "sni-qt/5139" WARN  13:46:05.401 void StatusNotifierItemFactory::connectToSnw() Invalid interface to SNW_SERVICE 

When I insert a USB flash drive I get following error on terminal and applet stops:

> terminate called after throwing an instance of 'std::invalid_argument'
>  what():  parse error - 1 is not a number
> Aborted

This system:
> Description:	Linux Mint 17.2 Rafaela
> Release:	17.2

dkopecek, 2015-11-02:
Hi, please see https://github.com/dkopecek/usbguard-applet-qt/issues/8, there's a workaround for this problem. Let me know if it works for you.

Thanks for testing btw.!

dx486, 2015-11-02:
It worked. Thank you for quick reply and for your coding effort!

dkopecek, 2015-11-02:
Great. I'll close this as it's tracked in the usbguard-applet-qt repo.

closed 2015-11-02 by dkopecek