USBGuard/usbguard #45: "Allow" does not work

dx486, 2015-11-05:
I click on "allow" but device is not being allowed. 

Here is the debug log: http://pastebin.com/CyBMeRw

dkopecek, 2015-11-09:
Looks like the problem is this:
```
[2015-11-05 21:39:38.155025] error: Exception: type must be number, but is object
[2015-11-05 21:39:38.155046] error: Invalid JSON object received. Disconnecting from the client.
```
I'm not yet sure why that happens. It could be related to the locale problem described in https://github.com/dkopecek/usbguard-applet-qt/issues/8#issuecomment-110770995.

I'll working on adding more and better debugging messages throughout the code.

dkopecek, 2016-01-09:
This should be resolved by 1f4d072d8ab9797bdb148982757d8ad3e2a46765. I was using the json API in a wrong way.

closed 2016-01-09 by dkopecek