USBGuard/usbguard #47: [RFE] Protected Mode is missing (GRSecurity support)

zaolin, 2016-01-26:
Hi,

currently the USB protection for bad usb only protects userspace attacks. Not kernelspace driver !
In order to support something like a protected mode (disabled usb stack) which can be enabled/disabled via qt applet.
I recommend to integrate the grsecurity DENY_USB feature which can be controlled via sysctl.
Pull request will follow in few days ;)

Regards Zaolin

dkopecek, 2016-01-26:
Hello and thanks! Looking forward to review the pull request.

dicer, 2017-10-12:
So currently if there is a bug in a kernel driver for some USB device and someone plugs in a USB device faking the usb id (and running an usb exploit), the kernel actually loads the driver already? No difference whether "block" or "reject" is used?

dkopecek, 2017-10-12:
@dicer The USB interface drivers are loaded after the authorization. The usbcore module is exposed to the USB devices before the authorization, so yes, if there's an exploitable bug in that module, then there's nothing to prevent the exploitation (maybe only some HW based solution..? I don't know). 

Anyway, if you manage to fake all the attributes needed to to recognize the device as an authorized one, then you are free to exploit anything that is in reach. Therefore you should probably write the (usbguard) rules as specific as possible for your use case to limit the attach surface as much as possible (i.e. don't use only the usb id to recognize authorized devices).

dicer, 2017-10-12:
Yeah, it's all about attack surface. Of course the usbcore needs to handle the basic communication with the device. I'm more worried about all the countless drivers which take over after that. As far as I know only the usb id and the interface descriptor matter, right? So for example, if I have no usb scanners allowed no one can exploit a bug in those interface drivers with usbguard blocking them? No difference between "block" and "reject" there?
On a hardened machine it makes sense to uninstall those modules anyway, but on a desktop machine that is not always feasible.

I also reread the description of this bug: This is more of a deny-all switch what gr_security has there, right? Is that even necessary with the current state of usbguard?

BenjaminSchubert, 2017-10-13:
GRSecurity is indeed a deny-all type of thing.

I'm not sure if embedding GRSec usb protection into usbguard makes sense, as GRSec is now behind a paywall and cannot be distributed, even Gentoo abandonned its support, but that's my opinion.

closed 2019-03-11 by dkopecek