USBGuard/usbguard #48: Gentoo Linux support

zaolin, 2016-01-26:
Hi,

I wrote for usbguard and the qt applet version specific gentoo packages. They can be found inside the stuge-overlay via https://gpo.zugaina.org/Overlays/stuge in few days.
It would be nice if you can update your README and add Gentoo as supported distribution.
I will reply on this thread asap if they are online.

dkopecek, 2016-01-26:
Hello. Thanks for this contribution. I don't have experience with how the overlays work in Gentoo, could you create a pull request with the changes to README.md that you have in mind? I'll happily merge them.

zaolin, 2016-01-29:
Gentoo packages and pull request are there. Maybe we see us at FOSDEM!
Good luck for your talk I am afraid that I have nearly no time to visit talks. ^^

dkopecek, 2016-02-08:
I think this one is resolved. Thanks again!

closed 2016-02-08 by dkopecek