USBGuard/usbguard #54: `usbguard list-devices --allowed` and `usbguard list-devices --blocked` have output swapped

genodeftest, 2016-02-12:
Steps to reproduce:
1. allow and/or deny some USB devices
2. restart usbguard systemd service (daemon)
3. `usbguard list-devices --allowed` lists only blocked devices
4. `usbguard list-devices --blocked` lists only allowed devices

I am running usbguard 0.5-1 from fedora Copr repos.

dkopecek, 2016-02-13:
Hi, thanks for the report! I think this bug should be fixed by af2a4825739ed69efbf75454d316e2b0210c68ad.

If that won't help, then I'll need the debug log from you. You can generate it by running usbguard-daemon with the debugging mode flag:
```
# usbguard-daemon -d -l /tmp/usbguard-daemon.log -c /etc/usbguard/usbguard-daemon.conf
```

closed 2016-02-22 by dkopecek

genodeftest, 2016-02-22:
Confirming fix in 0.5-3 from Fedora repos.
Thank you very much!

commit 96c7d9eb (2016-02-22): Use correct short option values for usbguard list-devices --allowed/--blocked options

Resolves: #54