USBGuard/usbguard #55: [RFE] Shuffle the allow, block and reject buttons

dkopecek, 2016-02-13:
_From @dkopecek on February 7, 2016 10:27_

Shuffling the order of the buttons could help in a situation where an attacker has some sort of automated control over an already authorized pointing device (e.g. a mouse, or even a wireless one -- in that case it's possible to operate it remotely by sending commands using the wireless protocol). If the buttons are shuffled, the attacker can't automate the attack using known button positions.

The feature should be configurable in the interface.

_Copied from original issue: dkopecek/usbguard-applet-qt#9_

rebecca-palmer, 2016-06-24:
It might be easier and/or more user-friendly to randomize the overall window position instead.

dkopecek, 2016-07-29:
The referenced commit adds and initial implementation of the position randomization (enabled by default). However, it needs testing and probably some additional work as it seems that it won't work for all environments. I'm using i3 (an older release) and it doesn't work. Probably because i3 overrides the window position when switching the window to a floating type -- just guessing.

dkopecek, 2016-08-06:
Seems to be working with newer i3 and also in KDE 5.

closed 2016-08-06 by dkopecek

indolering, 2017-02-17:
In what threat model does an attacker have control of the pointer but no control of the keyboard input?