USBGuard/usbguard #56: Applet failing with "Parse error - 1 is not a number"

dkopecek, 2016-02-13:
_From @rexcze on May 27, 2015 8:41_

I installed usbguard with applet from copr repozitory on Fedora 22 x86_64 system, than created rule wich allows only Mass storage devices, opened applet and inserted USB Flash disk. Than applet fails with this message:

 # usbguard-applet-qt 
> terminate called after throwing an instance of 'std::invalid_argument'
>  what():  parse error - 1 is not a number
> Neúspěšně ukončen (SIGABRT) (core dumped [obraz paměti uložen])

I can provide core dump if needed (as far as I can not upload it to Fedora bugzilla with automated abrt)

_Copied from original issue: dkopecek/usbguard-applet-qt#8_

dkopecek, 2016-02-13:
Thanks for the report! Yeah, the applet didn't passed the review process yet in Fedora, so there's no bugzilla component for it. Anyway, if you can provide the core dump or a backtrace, then put it somewhere where I can access it...

dkopecek, 2016-02-13:
_From @rexcze on May 27, 2015 12:20_

http://leteckaposta.cz/333845983

dkopecek, 2016-02-13:
Looks like this is a locale and nlohmann/json related problem. See https://github.com/nlohmann/json/issues/82. I'll try to workaround this somehow, until this is resolved in the json library.

dkopecek, 2016-02-13:
Hmm, looks like the commit doesn't help. However, you can workaround this from the command line:

     $ LC_NUMERIC=C ./usbguard-applet-qt

dkopecek, 2016-02-13:
_From @rexcze on June 22, 2015 8:23_

This workaround works. Thanks

genodeftest, 2016-07-12:
I think this issue is fixed with 0.5.10.

dkopecek, 2016-07-12:
Thanks, closing.

closed 2016-07-12 by dkopecek