USBGuard/usbguard #66: [Copr] Fedora 24 package broken due to libsodium soname bump

genodeftest, 2016-03-30:
I currently cannot install usbguard 0.5 from [your copr](https://copr.fedorainfracloud.org/coprs/mildew/usbguard/) due to a soname version bump in libsodium. usbguard requires libsodium.so.13 (which is available in F23), but libsodium on F24 was bumped to libsodium.so.18.

dkopecek, 2016-03-30:
Thanks for letting me know. I've submitted a new build: https://copr.fedorainfracloud.org/coprs/mildew/usbguard/build/172208/

dkopecek, 2016-03-30:
Hmm, looks like I forgot to update the BuildRequires. I'll submit another one in a minute.

dkopecek, 2016-03-30:
Unfortunately, there's a "cloud outage" at the moment and copr is not responding...

dkopecek, 2016-03-30:
https://copr.fedorainfracloud.org/coprs/mildew/usbguard/build/172212/

genodeftest, 2016-03-30:
Works fine, thank you!

closed 2016-03-30 by genodeftest