USBGuard/usbguard #67: [RFE] Add support for USB per-interface authorization

dkopecek, 2016-04-07:
per-interface authorization was recently added to the kernel, so we should reflect this in USBGuard too.

https://www.spinics.net/lists/linux-usb/msg126182.html

dkopecek, 2016-07-31:
# Rule language extension proposal to support per-interface policies:

To support per-interface policies, the rule language will be extended with per-interface rules which will be associated with a device rule using the "interface-policy" attribute. The attribute should have the following syntax:

`interface-policy <implicit-target> { interface-rules }`

The implicit target should be either allow or block. It will be used as a target for any interface that won't match a rule in the interface policy.

An interface rule will have similar syntax to that of a device rule. The differences will be in the device id syntax and in the set of available attributes. Instead of a USB device ID it will accept an USB interface type (numerical form). The attribute set should be the following:
 * index <num> ... index of the interface relative to the parent configuration
 * configuration <num> ... the configuration number of the interface
 * hash "<string>" ... hash of the interface and associated endpoint descriptors
 * rule conditions?

## Examples
```
allow 1234:1234 interface-policy block {
 allow 12:34:56 index 12 configuration 0 hash "1234567890abcdef"
}
```

## Known issues
 * Support for per-interface authorization in the Linux Kernel is "quite" new. How to handle that? Presence of this feature is easy to detect.
 * The rule language parser isn't designed to support multi-line rules. The rules can be quite long even now and with per-interface rules it will get even worse.

muelli, 2020-01-09:
FTR: This is how usbauth does it: https://github.com/kochstefan/usbauth-all/blob/master/usbauth/data/usbauth.conf

charlesgoyard, 2024-07-01:
While not fine-grained, a simple intermediate syntax like this could help meanwhile:

```
allow-interface { 03:01:01 03:01:00 } # yes to keyboards and mice
allow-interface { 07:*:* } # yes to the printing part of the printer, but not the included USB flash card port
```

In a place where I worked (3D work mostly), users tended to bring their mouse that presented all sorts of extensions in terms of interfaces, and we needed to maintain a bunch of exceptions. Having simple filtering at the interface level would help a lot.