USBGuard/usbguard #68: The usbguard Daemon won't start at booting

amenophobis, 2016-05-19:
Hello,

I've installed usbguard from the copr repo on fedora 23... everything works fine, but when I enable the service in Systemd it don't start at boot...

everytime I need to start the service manually...

does anyone have a solution for that problem?

genodeftest, 2016-05-20:
I don't have this problem. What's the output of `$ systemctl status usbguard.service` after boot but before starting it manually?

Mine:
```
● usbguard.service - USBGuard daemon
   Loaded: loaded (/usr/lib/systemd/system/usbguard.service; enabled; vendor preset: 
   Active: inactive (dead)
```
On the second line it reads "enabled". If yours doesn't, it is not enabled. If it failed, you should see an error message.

dkopecek, 2016-05-20:
@amenophobis Thanks for the report. Could you please share the output of systemctl status as mentioned in the previous comment and also the output of
```
$ sudo journalctl -u usbguard.service -r
```

amenophobis, 2016-05-20:
The output:

 systemctl status usbguard.service

● usbguard.service - USBGuard daemon
   Loaded: loaded (/usr/lib/systemd/system/usbguard.service; enabled; vendor preset: disabled)
   Active: inactive (dead)

> Vendor preset not good?

@dkopecek
The log is on my Computer at the moment not very useful because I started the service with a cronjob for root at boot... this has worked!!!

I have turned off the cronjob and send you the useful log today...

thanks for help...

dkopecek, 2016-05-20:
@amenophobis I reproduced the issue. The problem seems to be in the Install section of the unit. The `WantedBy=` value should be `basic.target` instead of `base.target`.

dkopecek, 2016-05-20:
@genodeftest Is it working for you with `WantedBy=base.target`? I wonder what could be the real problem. I don't really remember how I got to the "base.target" value as `systemd.special(7)` doesn't mention it.

amenophobis, 2016-05-20:
I changed it... works perfect...

closed 2016-05-20 by dkopecek

genodeftest, 2016-05-21:
Works fine for me with `WantedBy=base.target`. Might depend on systemd version, mine is 229-7 on Fedora 24 Beta. Changing it to `WantedBy=basic.target` seems to be the right way to do though.

commit 5775a957 (2016-05-20): We should be using basic.target in the systemd unit Install section

The base.target is incorrect and causes usbguard not to start at
boot.

Resolves: #68