USBGuard/usbguard #75: configure switches have no effect

murinicanor, 2016-05-27:
hi,

apparently the configure switches
--without-bundled-json and
--without-bundled-spdlog
have no effect. even if i use them, i get the following summary:

<pre>
## Libraries
    
    libudev: system-wide;  -ludev
      libqb: system-wide;  -lqb -ldl -lrt -lpthread
  libsodium: system-wide;  -lsodium
 libseccomp: system-wide; -lseccomp
  libcap-ng: system-wide; -lcap-ng
       json: bundled; -I$(top_srcdir)/src/ThirdParty/json/src
     spdlog: bundled; -I$(top_srcdir)/src/ThirdParty/spdlog/include
</pre>

also it would be great to have switches to disable bundled json and usbmon

dkopecek, 2016-05-28:
Hello, thanks, you're right. The options behave incorrectly.

w.r.t. usbmon: I think this one will have to stay bundled for a while. It doesn't provide a library to link to, so I have to include the sources. Anyway, at this moment the usbmon code isn't used because I/O monitoring isn't implement yet. If that is an issue for you (I guess because of packaging guidelines), then I can strip it off and move into a feature branch until the feature is complete.

closed 2016-05-28 by dkopecek

murinicanor, 2016-05-28:
thanks a lot for the fix!

sorry, there was a typo in my bugreport, i wanted to say, that it would be great to have a switch to disable bundled *quex* instead of json (because the json switch already existed...)

> w.r.t. usbmon: I think this one will have to stay bundled for a while. It doesn't provide a library to link to, so I have to include the sources. Anyway, at this moment the usbmon code isn't used [...]

yes, i'm working on a debian package. its my first package that i try to create according to the debian packaging guidelines. i plan to package quex and json and i also planned on packaging usbmon (i just would have tried to package https://github.com/radosroka/usbmon), but if its not needed yet, i'll just remove the folder

dkopecek, 2016-05-28:
@murinicanor I think unbundling quex is not worth the effort. I'd like to get rid of the quex and lemon generated code and replace both with a parser based on https://github.com/ColinH/PEGTL. See issue #65.

Thanks for the packaging work! I appreciate that a lot.

commit 0afc46a2 (2016-05-28): Fix behaviour of the --with-bundled-{json,spdlog} options

Resolves: #75