USBGuard/usbguard #78: Files missing copyright notice

murinicanor, 2016-05-29:
hi,

there are both files with copyright notices/license and files without one; it would be great if you could add a proper copyright notice also to these file (list is based on the 0.4 release)
<pre>
src/CLI/usbguard-append-rule.hpp
src/CLI/usbguard-allow-device.cpp
src/CLI/usbguard-watch.hpp
src/CLI/usbguard-watch.cpp
src/CLI/usbguard.hpp
src/CLI/usbguard-block-device.cpp
src/CLI/usbguard.cpp
src/CLI/usbguard-remove-rule.hpp
src/CLI/usbguard-generate-policy.hpp
src/CLI/usbguard-list-rules.cpp
src/CLI/usbguard-list-devices.cpp
src/CLI/IPCSignalWatcher.hpp
src/CLI/usbguard-list-devices.hpp
src/CLI/usbguard-rule-parser.cpp
src/CLI/usbguard-list-rules.hpp
src/CLI/usbguard-reject-device.cpp
src/CLI/usbguard-block-device.hpp
src/CLI/usbguard-allow-device.hpp
src/CLI/usbguard-append-rule.cpp
src/CLI/usbguard-reject-device.hpp
src/CLI/usbguard-remove-rule.cpp
src/CLI/IPCSignalWatcher.cpp
src/Library/RuleEvaluatedCondition.cpp
src/Library/LocaltimeCondition.cpp
src/Library/RuleAppliedCondition.hpp
src/Library/AllowedMatchesCondition.cpp
src/Library/RuleParser/Parser.h
src/Library/RuleParser/README.md
src/Library/RuleParser/Parser.y
src/Library/RuleParser/Lexer-token.hpp
src/Library/RuleParser/Lexer.cpp
src/Library/FixedStateCondition.hpp
src/Library/LocaltimeCondition.hpp
src/Library/RuleEvaluatedCondition.hpp
src/Library/DeviceManager.cpp
src/Library/RuleAppliedCondition.cpp
src/Library/RuleCondition.cpp
src/Library/RuleCondition.hpp
src/Library/IPCPrivate.hpp
src/Library/AllowedMatchesCondition.hpp
src/Library/FixedStateCondition.cpp
src/Library/RandomStateCondition.cpp
src/Library/RandomStateCondition.hpp
src/Library/IPCPrivate.cpp
src/Tests/main.cpp
src/Tests/Makefile.am
src/Tests/Unit/test_Rule.cpp
src/Tests/Unit/test_RuleParser.cpp
src/Tests/Regression/test_Rule_ghi37.cpp
</pre>

dkopecek, 2016-05-29:
Thanks for the report, I'll fix that for the next release.

closed 2016-05-30 by dkopecek

dkopecek, 2016-05-30:
I've added the copyright headers to files where I found it was missing. I haven't added the header to files which are generated by Quex and lemon (lexer and parser code). I'm not sure how the licensing work in this case. It will become irrelevant after I rewrite the parser using PETGL.

I've added a file copyright check to the test suite (which excludes the generated stuff), so that the presence of the copyright header in files is checked continuously.

murinicanor, 2016-05-31:
> I've added the copyright headers to files where I found it was missing. I haven't added the header to files which are generated by Quex and lemon (lexer and parser code). I'm not sure how the licensing work in this case. It will become irrelevant after I rewrite the parser using PETGL.

thanks!

> I've added a file copyright check to the test suite (which excludes the generated stuff), so that the presence of the copyright header in files is checked continuously.

wow, thanks! that makes it a lot easier, especially for debian!

commit eb7413bd (2016-05-30): Add copyright to files where it was missing

Resolves: #78