USBGuard/usbguard #82: [RFE] Expand USBGuard to work with optical media

redhatrises, 2016-06-02:
While this is most likely out of scope of this project, it would be really, really nice to expand USBGuard to work with optical media (CD/DVD/BluRay) devices as well. Not sure if optical media was considered at all, but expanding to encompass optical media would be a big win for users that need to lock down both optical and usb access. Maybe a new project called MediaGuard that works with usb and optical media?

dkopecek, 2016-06-03:
Hi. Thanks for this proposal. Optical media wasn't considered and it really is outside the scope of this project. However, I'm thinking about expanding the scope to PCI devices. Maybe something called DeviceGuard that would have bigger scope. It could try to handle/watch multiple subsystems incl. USB, PCI, optical media, displays... At this point it is really only an idea, probably not feasible without some additional support in the Linux kernel.

redhatrises, 2016-06-03:
> Hi. Thanks for this proposal. Optical media wasn't considered and it really is outside the scope of this project.

I guessed it would be. Wanted to see if it was on anyone's radar. 

> However, I'm thinking about expanding the scope to PCI devices. Maybe something called DeviceGuard that would have bigger scope.

+1. Expanding to more than just USB would be great! From the security realm, this helps to solve a multitude of issues especially addressing some NIST SP 800-53/DISA STIG controls.

> It could try to handle/watch multiple subsystems incl. USB, PCI, optical media, displays... 

That would be nice. Maybe start small with usb and optical media first?