USBGuard/usbguard #84: Sign releases (and maybe commits and tags)

murinicanor, 2016-06-11:
It would be great if you could gpg-sign your releases (https://wiki.debian.org/Creating%20signed%20GitHub%20releases) - i could then use the debian watch file to do signature verification (https://wiki.debian.org/debian/watch#Cryptographic_signature_verification)

dkopecek, 2016-06-17:
Hi and thanks for this suggestion. I'll start signing releases starting with the next release (0.5). I'll sign also tags (but so far I'm tagging only the releases...). I'll leave this open until the 0.5 release.

dkopecek, 2016-07-06:
Started signing commits since 9ea57056dc060dce9de4344629fe63cb28732ec9. My PGP keyid is [B25991BC](https://keys.fedoraproject.org/pks/lookup?search=0xB25991BC&fingerprint=on&hash=on&op=index).

dkopecek, 2016-07-08:
Released 0.5.10 including a signed tag and tarball.

closed 2016-07-08 by dkopecek