USBGuard/usbguard #86: Document that some systems have unstable bus numbering, breaking port-specific rules

rebecca-palmer, 2016-06-24:
My system's top level USB bus numbers often change on reboot (the same physical port may be 1-x or 3-x), which causes USBGuard to block devices that should have been allowed by a port-specific rule.

This can be worked around by not using port-specific rules, but as the Qt applet only generates such rules, users can easily be using them without knowing it.

We could theoretically fix this by adding a with-parent-hash to be used instead of with-port, but I don't know if it is common enough to be worth it.

dkopecek, 2016-06-25:
Hello, I think the "with-parent-hash" is a good idea and it shouldn't be that hard to implement. I'll try to implement that and also adjust the Qt applet behavior.

dkopecek, 2016-08-07:
Related commit 628d8aca274bc9a5a543689b542cf9fd65f3a369 introduced the DeviceRulesWithPort configuration option for the usbguard-daemon. Set to false by default so that the generated rules don't contain the via-port attribute. In case of the allowDevice action, the generated rules do contain the parent-hash attribute.

commit 7c38fa7d (2016-08-07): Update Rule Language documentation

Resolves: #86
Resolves: #87
closed 2016-08-07 by dkopecek