USBGuard/usbguard #87: GUI is unclear about: What's the difference between "Reject" and "Block"

genodeftest, 2016-07-12:
To me as a user it is quite unclear what the difference between "reject" and "block" is. There is no way to guess that from the GUI.

`man 5 usbguard-rules.conf` states:
```
• block -- deauthorize the device
• reject -- remove the device from the system
```
This is better, but still hard to understand for non-technical users. How about
* block: don't talk to this device for now
* reject: ignore this device as if didn't exist

In both cases the device doesn't show up in lsusb or any list provided by udisks.

Btw: what about power over USB?  Do blocked or rejected devices still get power? How about newer hardware which has to negotiate power consumption with the host?

From one USB flash disk I tested:
* block: device is still powered, LED is on
* reject: device LED is turned off

A smartphone I tested:
* block: device still powered, smartphone still asks me whether to act as flash disk or something else
* reject: smartphone still gets powered

If the difference between blocking and rejecting the device is too small, it might be useful to provide just one of both on the GUI.

dkopecek, 2016-07-12:
> • block -- deauthorize the device
> • reject -- remove the device from the system
> 
> This is better, but still hard to understand for non-technical users. How about
> 
>     block: don't talk to this device for now
>     reject: ignore this device as if didn't exist

Thanks. I like these descriptions so I'll update the docs. Or create a PR if you have time for that.

> In both cases the device doesn't show up in lsusb or any list provided by udisks.

If you just block the device, it should still be visible in lsusb output. However, if you reject it, then the USB device node gets removed and it won't be visible anymore. It might still get power. That probably depends on the hardware and USB host drivers (how they handle the remove request).

In the udisks case, no device should be visible in either case because the block device is created after the interfaces on the USB device get active. And they don't unless you authorize it.

> If the difference between blocking and rejecting the device is too small, it might be useful to provide just one of both on the GUI.

You're probably right. From the UI perspective it's much better to have just allow and block because then it's clear what each action does. However, I'll probably change it in a way that allows the user to configure whether he sees the reject button or on. And by default it won't be visible. In the configuration dialog, there will be some space for a short explanation of the difference between block and reject.

dkopecek, 2016-07-29:
The Qt applet now defaults to just "Allow" and "Block".

commit 7c38fa7d (2016-08-07): Update Rule Language documentation

Resolves: #86
Resolves: #87
closed 2016-08-07 by dkopecek