USBGuard/usbguard #93: Ship a change log

murinicanor, 2016-07-13:
hi,

it would be great if the release archives contained a CHANGELOG file. i was researching possibilities of generating that from the github release notes, but nothing useful came up. i know its probably cumbersome to take care of two places containing changes, but if you agree that it should be part of the released archives, i'd happy to create an initial file.

dkopecek, 2016-07-13:
Hello.

Yes, I agree. Maybe I'll teach myself to keep the git history clean so that it's useful as a changelog by itself. I think that the current form of the release tags is stable now. That should help with doing diffs between consecutive releases.

It would be nice to have at least some level of automation w.r.t. changelog, i.e. maybe some script that helps to generate some raw form of the changelog from the git history which will then be reviewed and transformed into new entries in the shipped changelog.

Feel free to create the initial version. Thanks!

closed 2016-07-17 by dkopecek

murinicanor, 2016-08-13:
the changelog is not included in the released archive ;)
(i'm don't know how that archive is created, otherwise i'd create a patch ;))

dkopecek, 2016-08-13:
Oh crap... I'll try to fix that. Thanks!

dkopecek, 2016-08-13:
Should be fixed now. I did a forced update of the tags and updated the files in the release. Sorry, for the inconvenience!

PR #94 (2016-07-17): add initial changelog
fixes #93 
i used the format proposed on http://keepachangelog.com, althought i'm not sure if its not too much to divide the changes in subsections (changed|added|fixed|removed)
commit ed095f01 (2016-07-17): Merge pull request #94 from murinicanor/add-changelog

Add initial changelog

Format based on http://keepachangelog.com/

Resolves: #93