USBGuard/usbguard #108: Execute command on rule match

Sec42, 2016-08-12:
I would like like to lock my screen on appearance of new HID devices. To make it more generic it would probably be good if usbguard would support to run a command/shell script on rule/device match.

genodeftest, 2016-08-13:
Currently, you could write a piece of code that registers to the dbus events and runs any command. You have to write the code though…

dkopecek, 2016-08-13:
@Sec42 Hi! Implementing this feature at the rule language level is currently not a priority. However, as @genodeftest already said, there's a D-Bus interface and you can watch for the events from your session and execute scripts based on that. Checkout https://askubuntu.com/questions/150790/how-do-i-run-a-script-on-a-dbus-signal for a possible implementation via the dbus-monitor utility.

Another idea would be to modify the usbguard CLI to support this via the watch sub-command. This would be easy to implement. Let's see whether I manage to deliver this in 0.6.x.

elKaZe, 2016-09-13:
Maybe a timeout and then trigger a script.

dsoprea, 2021-06-24:
I think this'd be awesome, too. Upon authorizing a device, maybe running a command that maps certain devices to certain commands or maps certain devices to an autostart process that further process a config on the device.

For example:

- cameras or thumbdrives used exclusively for images starting a sync process
- backup devices automatically starting a backup process
- webcams automatically starting a record process

Athwale, 2024-02-10:
Any update on this?