USBGuard/usbguard #111: allow comments in rules file

Sec42, 2016-08-13:
Currently no comments are allowed in the rules files (e.g. lines starting with "#") - this should be allowed

dkopecek, 2016-08-13:
Thanks for the suggestion. This is certainly a nice to have feature.

ilf, 2016-11-11:
Having generated and edited a rule file, I naturally added comment lines, only to discover usbguard would fail starting. I would also like this. :)

dkopecek, 2016-11-12:
@ilf Hi and thanks for the feedback. I guess I'm going to implement this sooner than planned.

Dridi, 2016-12-29:
As a side note, empty lines seem to end the parsing prematurely, and *silently*. I noticed it while playing with `usbguard-rule-parser` after getting a parsing error on a wannabe comment.

So please support both comments and empty lines :)

Dridi, 2017-12-11:
To anyone interested, I did some testing on 0.7.1 because I initially thought that #209 solved this issue. Doing some testing with a bunch of devices and twisting my rules to make sure that I'm not weakening my setup I came to the conclusion that empty lines are indeed supported.

The `usbguard-rule-parser` program on the other hand doesn't use the same code to parse rules files, and silently ignores whatever comes after the first empty line.

This ignores empty lines:

https://github.com/USBGuard/usbguard/blob/6b7863aee923b91a2c50bd5078a458f85ecec453/src/Library/RuleSetPrivate.cpp#L87-L89

This stops on the first empty line:

https://github.com/USBGuard/usbguard/blob/e7f984bf1319df6059f09789c81cda877964ec24/src/CLI/usbguard-rule-parser.cpp#L104-L106

While we still don't have comments in rules files, I can easily identify most of my devices by simply separating them with empty lines (there's one device I couldn't identify easily that took some plugging and unplugging to figure out=).

bduff-sjm, 2018-04-03:
Also, it would be nice if it had the usual "rules.d" sort of directory, so /etc/usbguard/rules.d/*.conf would be processed in lexical order.

closed 2018-07-24 by dkopecek

jk-1, 2019-01-06:
Is this comment support already implemented? I tried v 0.7.2 and it does not have support for comments in rules.conf file.

Dridi, 2019-01-06:
Unfortunately https://github.com/USBGuard/usbguard/commit/09be6dcc1e2004b06d610278020b3468db69cc57 is not part of a release, it was pushed on July 24th and the last release was on July 12th.

I'm very much waiting for the next release to finally get comments in my setup. My concern is that the last code update on the master branch was on October :(

jk-1, 2019-01-07:
OK, hopefully it is coming soon, because long configuration files without any comments are extremely hard to maintain.