USBGuard/usbguard #114: Make `--help` indent better

genodeftest, 2016-08-14:
A perfectionistic comments on the `--help` "page":

Currently, `--help` does not indent text for subcommands and text:
```
       allow-device [OPTIONS] <id>

       Authorize a device identified by the device id to interact with the system.

       Available options:

       -p, --permanent
              Make  the  decision permanent.  A device specific allow rule will be
              appended to the current policy.

       -h, --help
              Show help.
```
I'd prefer it this way:
```
       allow-device [OPTIONS] <id>

           Authorize a device identified by the device id to interact with the system.

           Available options:

           -p, --permanent
               Make  the  decision permanent.  A device specific allow rule will be
               appended to the current policy.

           -h, --help
               Show help.
```
With this change you could omit the `~ ~ ~` lines. This is the way `man bash` and `man systemctl` do it.

dkopecek, 2016-08-14:
I hope that this will be doable with the markdown -> roff (via pandoc) conversion path. I'll adopt this format if it works. Thanks for the feedback!

dkopecek, 2016-08-16:
I tried to indent the sub-command description block in the markdown source file but it doesn't work :-( Any idea how to achieve this in markdown?

endomandi, 2016-12-20:
[0001-Indent-subcommand-blocks-as-subsections.patch.txt](https://github.com/dkopecek/usbguard/files/665036/0001-Indent-subcommand-blocks-as-subsections.patch.txt)

I've only checked the roff output.