USBGuard/usbguard #115: Don't collapse usb devices tree on detaching device

genodeftest, 2016-08-14:
Steps to reproduce:
1. open UsbGuard Qt Applet window (on "Devices" tab)
2. attach any usb device
3. block or accept the new device
4. detach the new device

What happens:
Device tree collapses

What is expected to happen:
Don't collapse device tree. (Only when a user clicked on the collapse button.)

Affected version:
usbguard-0.5.11-2.fc24.x86_64

dkopecek, 2016-08-14:
Right. The default state should be expanded, so it seems that it's not working.

dkopecek, 2016-08-15:
I've disabled the option to collapse items in the list. I'm too lazy to implement the proper tree view state handling. I'm going to refactor the applet code in the future (if nobody else will do that before me ;-) but for now it's not a priority.

genodeftest, 2016-08-15:
Looks like the better solution to me anyway. Who has 127 devices (that's the upper limit, at least for USB 1.0 through 2.0) anyway? And still scrolling is easier than collapsing/uncollapsing.

closed 2016-08-15 by genodeftest