USBGuard/usbguard #116: Don't show dialog twice for one device

genodeftest, 2016-08-14:
Sometimes when attaching an ordinary USB flash disk device I get two dialogs of USBGuard Qt Applet. When I click any button on the first dialog, the second dialog is still visible. The reason for having 2 dialogs probably is the fact that USB ports are not physically perfect allowing the device to hang a bit loose while it is being attached, resulting in a disconnect between two connects in way less than one second.

In this case I attached a USB flash device and got these logs:

USBGuard:
```
[So. Aug. 14 10:46:09 2016] Inserted: 1
[So. Aug. 14 10:46:09 2016] Blocked: 1
[So. Aug. 14 10:46:11 2016] Allowed: 1
[So. Aug. 14 10:46:13 2016] IPC call failed: allowDevice: NotFound: Requested action depends on a resource which is not available
```
The 4th line happened when I clicked "Allow" on the second dialog.

dmesg before allowing the device:
```
[ 1507.098088] usb 1-1.3: new high-speed USB device number 6 using ehci-pci
[ 1507.184207] usb 1-1.3: New USB device found, idVendor=abcd, idProduct=1234
[ 1507.184214] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1507.184218] usb 1-1.3: Product: 1
[ 1507.184221] usb 1-1.3: Manufacturer: 1
[ 1507.184224] usb 1-1.3: SerialNumber: Љ
[ 1507.184655] usb 1-1.3: Device is not authorized for usage
[ 1507.433447] usb 1-1.3: USB disconnect, device number 6
[ 1507.714037] usb 1-1.3: new high-speed USB device number 7 using ehci-pci
[ 1507.799887] usb 1-1.3: New USB device found, idVendor=abcd, idProduct=1234
[ 1507.799892] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1507.799895] usb 1-1.3: Product: 1
[ 1507.799898] usb 1-1.3: Manufacturer: 1
[ 1507.799900] usb 1-1.3: SerialNumber: Љ
[ 1507.800306] usb 1-1.3: Device is not authorized for usage
[ 1509.470664] usb-storage 1-1.3:1.0: USB Mass Storage device detected
[ 1509.470921] scsi host9: usb-storage 1-1.3:1.0
[ 1509.471309] usb 1-1.3: authorized to connect
[ 1510.475152] scsi 9:0:0:0: Direct-Access     General  UDisk            5.00 PQ: 0 ANSI: 2
[ 1510.476199] sd 9:0:0:0: Attached scsi generic sg6 type 0
[ 1510.477247] sd 9:0:0:0: [sdf] 7864320 512-byte logical blocks: (4.03 GB/3.75 GiB)
[ 1510.477877] sd 9:0:0:0: [sdf] Write Protect is off
[ 1510.477883] sd 9:0:0:0: [sdf] Mode Sense: 0b 00 00 08
[ 1510.478596] sd 9:0:0:0: [sdf] No Caching mode page found
[ 1510.478605] sd 9:0:0:0: [sdf] Assuming drive cache: write through
[ 1510.483609]  sdf:
[ 1510.486701] sd 9:0:0:0: [sdf] Attached SCSI removable disk
```
Where, at 1509, I allowed device access.

Idea for a workaround:
1. When a device is attached that has been attached previously in the last e.g. 2 seconds, check whether it is attached to the same port. If the device is at the same port, they are probably the same device. Show dialog only once in this case.
2. When a device is detached and was not yet allowed, change or destroy the dialog for it. There is no use in having dialogs shown for devices not attached to the computer any more.

dkopecek, 2016-08-14:
I'm seeing this too sometimes and it's annoying. I'll try to fix that in the near future, I guess it won't be hard to implement. Thanks for the feedback!

dkopecek, 2019-06-28:
Closing this as I've removed the applet from the main usbguard repository in https://github.com/USBGuard/usbguard/commit/bf3bb09edf491892617d35b55edec06f92d2743f. The plan is to simplify it down to a minimal required use case and maintain it in a separate repository.

closed 2019-06-28 by dkopecek