USBGuard/usbguard #117: Some USB devices are blocked by default, no dialog shows up

genodeftest, 2016-08-14:
Steps to reproduce:
1. attach a "Logitech USB Headset H340" (idVendor=046d, idProduct=0a38)
2. watch logs

What happens:
I don't get a dialog to ask for allowing or blocking this device.
USBGuard logs don't show anything about this device at all.
dmesg:
```
[ 2457.095548] usb 1-1.3: new full-speed USB device number 10 using ehci-pci
[ 2457.382677] usb 1-1.3: New USB device found, idVendor=046d, idProduct=0a38
[ 2457.382682] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 2457.382684] usb 1-1.3: Product: Logitech USB Headset H340
[ 2457.382686] usb 1-1.3: Manufacturer: Logitech Inc.
[ 2457.383032] usb 1-1.3: Device is not authorized for usage
[ 2457.393384] usb 1-1.3: USB disconnect, device number 10
```
Journalctl reveals:
```
kernel: usb 1-1.3: new full-speed USB device number 10 using ehci-pci
kernel: usb 1-1.3: New USB device found, idVendor=046d, idProduct=0a38
kernel: usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
kernel: usb 1-1.3: Product: Logitech USB Headset H340
kernel: usb 1-1.3: Manufacturer: Logitech Inc.
kernel: usb 1-1.3: Device is not authorized for usage
usbguard-daemon[6129]: [2016-08-14 11:01:59.152151] error: Exception caught during device insertion processing: /sys/devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.3: Invalid descriptor data: bLength doesn't match expected bLenght for this descriptor type
kernel: usb 1-1.3: USB disconnect, device number 10
```

The device itself is working if I `systemctl stop usbguard.service`. In this case I can even pass it through to a virtual machine client where it works perfectly.

Affected version:
usbguard-0.5.11-2.fc24.x86_64

I don't know whether this is a bug in USBGuard or in the device itself, this is up to you to decide. But it would be helpful if the logs in USBGuard Qt Applet would show all the error and warning messages .

dkopecek, 2016-08-14:
This is interesting. Could you please send me the USB descriptor data? You can use [this script](https://raw.githubusercontent.com/dkopecek/usbguard/master/scripts/usb-descriptor-collect.sh) to collect the data. Note that the script will collect all descriptors.

If this is not a bug in usbguard, then I'll have to probably implement an option or some way to accept even such devices -- that will be a PITA :smile:

w.r.t the logs/messages, you are right. They shouldn't be hidden. That just confuses users.

genodeftest, 2016-08-14:
Logs send to your email address from GPG key.

PS: You probably may want to change the hashing algorithm in your script to use SHA-256 as usbguard did.
Thanks for the great work!

dkopecek, 2016-08-15:
This is bug in the USB descriptor parser. The problem was in that usbguard didn't recognize a special type of an endpoint descriptor (audio device related) which has the 0x05 type but a length of 9 bytes. So, there are two types of endpoint descriptors. One has a fixed length of 7 and the other of 9 bytes.

There are some descriptor types that usbguard still doesn't recognize but it just skips analyzing the content for them. These do not trigger the "invalid data" exception. However, for known types, there's an assumption that any deviation from expected values means that the descriptor is not valid and could be a potential exploitation payload.

I've added support for the audio device related endpoints and I'll keep adding support for other types as well.

closed 2016-08-15 by dkopecek

DarkPark, 2017-02-19:
in general it's a very good project but at the moment unfortunately unusable
I've tried all my devices and all were blocked silently without any questions from applet
* 4 different flash drives
* zalman external virtual cd
* creative sound card
* xiaomi mi5 phone
* pocketbook e-reader

dkopecek, 2017-02-19:
@DarkPark thanks for the feedback. Some questions:

- What release are you using?
- How does your usbguard-daemon configuration and the policy look like?
- Did you enable the user, under which you are running the applet, to communicate via the IPC. Tthis is done in the usbguard-daemon configuration file, look for IPCAllowedUsers.

You can use the CLI to try to list and authorize (allow) devices:

1. List devices
```
> usbguard list-devices
```
2. Allow a particular device  (the id is the first number in the `list-devices` command output)
```
> usbguard allow-device <id>
```

DarkPark, 2017-02-19:
thank you for quick and very helpful reply
point with IPCAllowedUsers I haven't found on the main page
now it works (though I recheck only some devices)

commit 18e3f35b (2016-08-15): Add support USB endpoint descriptor of length 9

 - refactored USB parser to support multiple handlers per
   USB descriptor type
 - added support for an audio device related endpoint descriptor
 - successfully tested with 141 samples of USB descriptor data

Resolves: #117