USBGuard/usbguard #122: lockout on a PC during system upgrade

genodeftest, 2016-08-19:
Some minutes ago I ran into a pretty serious bug in usbguard: I was locked out of my PC during a system upgrade.

I updated these packages:
* dnf
* evolution-data-server
* glib2
* glibc
* gtk3
* kernel (including headers, modules, tools)
* lcms2
* util-linux (including libblkid, libmount, libsmartcols, libuuid)
* selinux 
* systemd

Through one of these updates, something changed in devfs so that usbguard treated my USB root hubs as if they were attached during this update. Since in my case usbguard is configured to block new devices by default, all USB devices were blocked, including mouse and keyboard. This is pretty bad.

I'll try hunting down which package update triggered this but I wanted to post it here first.

I'm running usbguard 0.5.14 on Fedora 24 with updates-testing repos enabled.

dkopecek, 2016-08-19:
So you have no rules in the policy, just an implicit block target configured in usbguard-daemon.conf? What about the PresentDevicePolicy and PresentControllerPolicy settings? I guess it was the systemd package update which triggered that, but I haven't tried to reproduce this yet.

genodeftest, 2016-08-19:
Some additional files:
* output of [`dnf history info last`](https://github.com/dkopecek/usbguard/files/427440/dnf.history.info.last.txt)
* [`journalctl --since="2016-08-19 17:49:00" --until="2016-08-19 17:49:30" > /tmp/journal-excerpt.txt`](https://github.com/dkopecek/usbguard/files/427446/journal-excerpt.txt)

I'm not 100% sure this isn't a bug in systemd-udev or util-linux, because after a look at the syslog it looks like something else is wrong.

genodeftest, 2016-08-19:
> So you have no rules in the policy, just an implicit block target configured in usbguard-daemon.conf?

Yes. I probably should add one some time.

> What about the PresentDevicePolicy and PresentControllerPolicy settings?

```
PresentDevicePolicy=keep
PresentControllerPolicy=keep
```

Both configurations were "violated" here.

dkopecek, 2016-08-19:
The Present*Policy settings currently apply only when the daemon starts. I guess I'll have to either add some logic to recognize present devices even for the "add" UDev event or ensure that the daemon will be restarted with udevd.

Another option would be to bypass udev altogether, listening to the uevents from kernel directly. That would remove the "udev point of failure" from the picture.

I'll look into it over the weekend.

genodeftest, 2016-08-19:
Ok, I can confirm this issue is triggered when the systemd-udev package gets upgraded. This seems to be happening no matter the version of systemd-udev.

commit f2d4a02f (2016-12-15): Extend usbguard-daemon configuration API

  - Added InsertedDevicePolicy configuration option to control the policy
    method for inserted devices.
  - Added RestoreControllerDeviceState configuration option.
  - Added DeviceManagerBackend configuration option. This option can be used
    to select from several device manager backend implementations.
  - Implemented an uevent based device manager backend.
  - Added setParameter, getParameter IPC (incl. D-Bus) methods.
  - Added set-parameter, get-parameter CLI subcommands.
  - Qt Applet: Added Spanish (es_AR) translation.
  - Removed UDev bas [... cut]
commit a0f68d81 (2016-12-20): Extend usbguard-daemon configuration API

  - Added InsertedDevicePolicy configuration option to control the policy
    method for inserted devices.
  - Added RestoreControllerDeviceState configuration option.
  - Added DeviceManagerBackend configuration option. This option can be used
    to select from several device manager backend implementations.
  - Implemented an uevent based device manager backend.
  - Added setParameter, getParameter IPC (incl. D-Bus) methods.
  - Added set-parameter, get-parameter CLI subcommands.
  - Qt Applet: Added Spanish (es_AR) translation.
  - Removed UDev bas [... cut]
commit 66db5358 (2016-12-20): Extend usbguard-daemon configuration API

  - Added InsertedDevicePolicy configuration option to control the policy
    method for inserted devices.
  - Added RestoreControllerDeviceState configuration option.
  - Added DeviceManagerBackend configuration option. This option can be used
    to select from several device manager backend implementations.
  - Implemented an uevent based device manager backend.
  - Added setParameter, getParameter IPC (incl. D-Bus) methods.
  - Added set-parameter, get-parameter CLI subcommands.
  - Qt Applet: Added Spanish (es_AR) translation.
  - Removed UDev bas [... cut]
commit 34b3c7fc (2016-12-20): Extend usbguard-daemon configuration API

  - Added InsertedDevicePolicy configuration option to control the policy
    method for inserted devices.
  - Added RestoreControllerDeviceState configuration option.
  - Added DeviceManagerBackend configuration option. This option can be used
    to select from several device manager backend implementations.
  - Implemented an uevent based device manager backend.
  - Added setParameter, getParameter IPC (incl. D-Bus) methods.
  - Added set-parameter, get-parameter CLI subcommands.
  - Qt Applet: Added Spanish (es_AR) translation.
  - Removed UDev bas [... cut]
closed 2016-12-29 by dkopecek