USBGuard/usbguard #123: Issue with applet not opening windows

phocean, 2016-08-20:
There is an issue with the applet on Fedora 24 / Gnome 3.20.

Clicking on the tray icon opens the window only once.
Subsequent attempts to reopen it by clicking on this icon results in a quick open/close event (I can see it in the dash, but no window ever appears).

No console error is output.

PS: Wonderful tool, thanks a lot!

dkopecek, 2016-08-23:
Hi and thanks for the report! As I'm not a Gnome desktop user, I haven't noticed this. Anyway, I'll try to reproduce this in a VM.

phocean, 2016-08-27:
I am not using Wayland. Fedora 24 is still using Xorg by default.

dkopecek, 2016-09-13:
Please try the latest Copr build. The issue should be addressed by commit c1f064a.

phocean, 2016-09-14:
I just updated to usbguard-applet-qt.x86_64 0.6.1-0.1.1473773926.fc24 and it is actually worse than the previous version.

The previous version partially fixed the issue, in a way that simple click on the tray icon would not work, but double click did open the window correctly.

Now, following today's update, it does not work in both cases.

dkopecek, 2016-09-14:
Ok, this one should definitely fix the issue. I've tested it in Gnome and it works for me. Copr builds should be available shortly.

phocean, 2016-09-15:
The last update is perfect, no more issue at all.
Thank you very much !!

closed 2016-09-15 by phocean

dkopecek, 2016-09-15:
Great, thanks for testing!