USBGuard/usbguard #124: Daemon crashes on update while usbguard-applet-qt is running

genodeftest, 2016-09-02:
I just updated usbguard{-applet-qt} from 0.5.12-1.fc24 to 0.6.0-0.1.1472764159.fc24. I was running an online update, so my desktop session with usbguard-applet-qt was still running.

I got these lines in systemd-journal repeating every 6 seconds:
```
 usbguard-daemon[3701]: [2016-09-02 09:07:52.668727] error: Exception: Unknown IPC payload type
 usbguard-daemon[3701]: [2016-09-02 09:07:52.668981] error: Invalid JSON object received. Disconnecting from the client.
 usbguard-daemon[3701]: usbguard-daemon: ipcs.c:579: qb_ipcs_connection_unref: Assertion `0' failed.
 systemd[1]: usbguard.service: Main process exited, code=killed, status=6/ABRT
 systemd[1]: usbguard.service: Unit entered failed state.
 systemd[1]: usbguard.service: Failed with result 'signal'.
 systemd[1]: usbguard.service: Service hold-off time over, scheduling restart.
 ```

usbguard probably needs to handle updates in a better way. Currently, the update forces a restart of the daemon, but not the applet (or the CLI client I think). In my opinion it should either do both (which is kinda hard to realize) or nothing (which is the way most package updates work on Fedora). So I'd prefer the latter if it is the same for you.

dkopecek, 2016-09-02:
Thanks for the report. I'll remove the restart from the %postun section, i.e. use just systemd_postun in there instead of systemd_postun_with_restart.

dkopecek, 2016-09-02:
The crashing of the daemon should be resolved with the latest commits. I've submitted a new testing build of the master branch version to Copr, should be available shortly.

closed 2016-09-02 by dkopecek

genodeftest, 2016-09-03:
Crash is gone now, thanks!

Still, it looks like the daemon is restarting since the applet lost connection at least twice during upgrade. This could have been a postun-script from the old version though. Now a `dnf reinstall usbguard*` did work without any issues.