USBGuard/usbguard #128: usbguard generate-policy does not produce any output

murinicanor, 2016-09-08:
hi,

i'm using 0.6 on debian for some days now and i just tried to use `usbguard generate-policy`. but it does not produce any output.
(i thought about using it to create an initial policy on first install on debian)

genodeftest, 2016-09-08:
No output at all? That's pretty weird. Are you sure you didn't accidentially alias usbguard? What does `which usbguard` show? Or `usbguard --help`? `lsusb`?

murinicanor, 2016-09-08:
> Are you sure you didn't accidentially alias usbguard? What does which usbguard show? 

/usr/bin/usbguard

and it is a:
`/usr/bin/usbguard: ELF 64-bit LSB shared object, x86-64, version 1 (GNU/Linux), dynamically linked, interpreter /lib64/ld-linux-x86-64.so.2, for GNU/Linux 2.6.32, BuildID[sha1]=e968a6d22ef96e27cc5f49722cdb76bc1d42f066, stripped`

> Or usbguard --help? 

`[2016-09-08 19:54:15.121794] error: Unknown subcommand: --help
 Usage: usbguard [OPTIONS] <command> [COMMAND OPTIONS] ...
 [...]
`
and the help text follows

> lsusb?

<pre>Bus 001 Device 005: ID 05c8:0374 Cheng Uei Precision Industry Co., Ltd (Foxlink) 
Bus 001 Device 004: ID 03f0:581d Hewlett-Packard 
Bus 001 Device 008: ID 1050:0407 Yubico.com 
Bus 001 Device 003: ID 0424:2134 Standard Microsystems Corp. Hub
Bus 001 Device 002: ID 8087:8001 Intel Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub</pre>

genodeftest, 2016-09-08:
It might be possible that usbguard isn't detecting some of your hardware. I had the same issue (see #117) with just one piece of hardware which was blocked by default due to a preference I made.

You might want to try sending @dkopecek the USB descriptor data as he wrote in a comment to that bug:
> This is interesting. Could you please send me the USB descriptor data? You can use [this script](https://raw.githubusercontent.com/dkopecek/usbguard/master/scripts/usb-descriptor-collect.sh) to collect the data. Note that the script will collect all descriptors.

So you might want to send the data via mail to him instead of posting it here.

murinicanor, 2016-09-08:
it works with 0.5.11 by the way, so it's probably a change since then

> It might be possible that usbguard isn't detecting some of your hardware. I had the same issue (see #117) with just one piece of hardware which was blocked by default due to a preference I made.
> 
> You might want to try sending @dkopecek the USB descriptor data

i'll definitly do that

commit bfeb3380 (2016-09-12): Adapt PolicyGenerator class to new DeviceManager API

Resolves: #128
closed 2016-09-12 by dkopecek