USBGuard/usbguard #130: usbguard-applet-qt always crashes when restarting daemon

genodeftest, 2016-09-08:
Steps to reproduce:
1. have usbguard.service and usbguard-applet-qt running
2. restart usbguard.service by `systemctl restart usbguard.service`

What happens:
Applet crashes. Backtrace:
```
#0  0x00007ffff4e526f5 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:54
#1  0x00007ffff4e542fa in __GI_abort () at abort.c:89
#2  0x00007ffff57913ed in __gnu_cxx::__verbose_terminate_handler() ()
    at ../../../../libstdc++-v3/libsupc++/vterminate.cc:95
#3  0x00007ffff578f196 in __cxxabiv1::__terminate(void (*)()) (handler=<optimized out>)
    at ../../../../libstdc++-v3/libsupc++/eh_terminate.cc:47
#4  0x00007ffff578f1e1 in std::terminate() () at ../../../../libstdc++-v3/libsupc++/eh_terminate.cc:57
#5  0x00007ffff7b51699 in usbguard::IPCClientPrivate::connect() (__t=<unknown type in /usr/lib/debug/usr/lib64/libusbguard.so.0.0.0.debug, CU 0x4ec5d, DIE 0x68ff6>, this=0x555555a2d030)
    at /usr/include/c++/6.1.1/thread:153
#6  0x00007ffff7b51699 in usbguard::IPCClientPrivate::connect() (this=0x555555a2d018)
    at src/Common/Thread.hpp:62
#7  0x00007ffff7b51699 in usbguard::IPCClientPrivate::connect() (this=0x555555a2cfa0)
    at src/Library/IPCClientPrivate.cpp:101
#8  0x00007ffff7b50a49 in usbguard::IPCClient::connect() (this=this@entry=0x7fffffffdc30)
    at src/Library/IPCClient.cpp:35
#9  0x0000555555564e58 in MainWindow::ipcTryConnect() (this=0x7fffffffdc00)
    at src/GUI.Qt/MainWindow.cpp:393
#10 0x0000555555572c65 in MainWindow::qt_static_metacall(QObject*, QMetaObject::Call, int, void**) (_o=<optimized out>, _c=<optimized out>, _id=<optimized out>, _a=<optimized out>)
    at src/GUI.Qt/MainWindow.moc.cpp:254
#11 0x00007ffff5cf2b92 in QMetaObject::activate(QObject*, int, int, void**) (sender=sender@entry=0x7fffffffdc78, signalOffset=<optimized out>, local_signal_index=local_signal_index@entry=0, argv=argv@entry=0x0) at kernel/qobject.cpp:3730
#12 0x00007ffff5cf33b7 in QMetaObject::activate(QObject*, QMetaObject const*, int, void**) (sender=sender@entry=0x7fffffffdc78, m=m@entry=0x7ffff5f03d80 <QTimer::staticMetaObject>, local_signal_index=local_signal_index@entry=0, argv=argv@entry=0x0) at kernel/qobject.cpp:3595
#13 0x00007ffff5d653f0 in QTimer::timeout(QTimer::QPrivateSignal) (this=this@entry=0x7fffffffdc78)
    at .moc/moc_qtimer.cpp:198
#14 0x00007ffff5cfef88 in QTimer::timerEvent(QTimerEvent*) (this=0x7fffffffdc78, e=<optimized out>)
    at kernel/qtimer.cpp:247
#15 0x00007ffff5cf3b8b in QObject::event(QEvent*) (this=0x7fffffffdc78, e=<optimized out>)
    at kernel/qobject.cpp:1278
#16 0x00007ffff655dc0c in QApplicationPrivate::notify_helper(QObject*, QEvent*) (this=<optimized out>, receiver=0x7fffffffdc78, e=0x7fffffffd930) at kernel/qapplication.cpp:3804
#17 0x00007ffff65630ef in QApplication::notify(QObject*, QEvent*) (this=
    0x7fffffffdbf0, receiver=0x7fffffffdc78, e=0x7fffffffd930) at kernel/qapplication.cpp:3561
#18 0x00007ffff5cca7aa in QCoreApplication::notifyInternal2(QObject*, QEvent*) (receiver=0x7fffffffdc78, event=event@entry=0x7fffffffd930) at kernel/qcoreapplication.cpp:1015
#19 0x00007ffff5d197ab in QTimerInfoList::activateTimers() (event=0x7fffffffd930, receiver=<optimized out>) at ../../src/corelib/kernel/qcoreapplication.h:225
#20 0x00007ffff5d197ab in QTimerInfoList::activateTimers() (this=0x5555558aa940)
    at kernel/qtimerinfo_unix.cpp:637
#21 0x00007ffff5d19d01 in timerSourceDispatch(GSource*, GSourceFunc, gpointer) (source=<optimized out>)
    at kernel/qeventdispatcher_glib.cpp:176
#22 0x00007ffff3e386ba in g_main_context_dispatch (context=0x7fffdc0016f0) at gmain.c:3154
#23 0x00007ffff3e386ba in g_main_context_dispatch (context=context@entry=0x7fffdc0016f0) at gmain.c:3769
#24 0x00007ffff3e38a70 in g_main_context_iterate (context=context@entry=0x7fffdc0016f0, block=block@entry=1, dispatch=dispatch@entry=1, self=<optimized out>) at gmain.c:3840
#25 0x00007ffff3e38b1c in g_main_context_iteration (context=0x7fffdc0016f0, may_block=may_block@entry=1)
    at gmain.c:3901
#26 0x00007ffff5d1a22f in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) (this=0x5555558aa6e0, flags=...) at kernel/qeventdispatcher_glib.cpp:417
#27 0x00007ffff5cc95ea in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) (this=this@entry=0x7fffffffdb80, flags=..., flags@entry=...) at kernel/qeventloop.cpp:204
#28 0x00007ffff5cd117c in QCoreApplication::exec() () at kernel/qcoreapplication.cpp:1285
#29 0x000055555555eed7 in main(int, char**) (argc=<optimized out>, argv=<optimized out>)
    at src/GUI.Qt/main.cpp:27
```

Full backtrace: https://gist.github.com/genodeftest/9fe173f58f55affd42db9cb13f204cea

Reproduced in versions 0.6.0-1.fc24 and 0.6.1-0.1.1473253488.fc24. Backtrace is created from latter versionrunning.

dkopecek, 2016-09-13:
This issue should be resolved in the latest Copr build (works for me), please test and close if it works for you.

genodeftest, 2016-09-13:
Thanks. Fixed in usbguard-0.6.1-0.1.1473773926.fc24.x86_64

closed 2016-09-13 by genodeftest