USBGuard/usbguard #132: soname version

murinicanor, 2016-09-10:
shouldn't there be soname version changes between 0.x and 0.y releases of libusbguard?

dkopecek, 2016-09-13:
Hi, I plan to start with maintaining the version once the project hits the first stable version 1.0.0. However, if you think it would be useful to start earlier with this, I'll reconsider original plan.

murinicanor, 2016-09-13:
it would help me with packaging for debian, because i'll have to provide different soname versions for the shared library package- but i can understand if it has nor priority at the moment. i can also try to find out how to set the soname myself while building the package (i guess the LT_* variables are what i'm looking for)

murinicanor, 2016-12-15:
hi, i've repackaged usbguard for debian and moved the shared library to a private dir (and removed the -dev package for now) so this is not an issue any more ;)

closed 2016-12-15 by murinicanor