USBGuard/usbguard #134: Translation

elKaZe, 2016-09-13:
I would like to translate usbguard applet to Spanish. Can you host strings/documentation files in a services such as Transiflex or something?

dkopecek, 2016-09-13:
Hello and thanks! I'm definitely interested in various translations of the UI and/or documentation. I know that Zanata (but never actually tried it). Would that work for you?

elKaZe, 2016-09-13:
I've help translating Tor and Diaspora on Transiflex, because of that I know it, but I don't have any problems whit Zanata, so if it's more comfortable to you, lets use it!

dkopecek, 2016-09-16:
I've completed the implementation of UI translations support. Maybe there won't be a need for using a service for this. I guess a pull-request that adds the Qt .ts file will suffice, what do you think? Please see [TRANSLATIONS.md](https://github.com/dkopecek/usbguard/blob/master/src/GUI.Qt/TRANSLATIONS.md) for details on how to create the .ts files for the applet.

elKaZe, 2016-09-16:
It's ok for the UI, i don't think so for the documentation.
Btw, i cant create the .ts file:
`lupdate-qt4 -source-language en_US -target-language es_AR -ts translations/es_AR.ts
Updating 'translations/es_AR.ts'...
    Found 0 source text(s) (0 new and 0 already existing)`

dkopecek, 2016-09-16:
> It's ok for the UI, i don't think so for the documentation.
> Btw, i cant create the .ts file:
> lupdate-qt4 -source-language en_US -target-language es_AR -ts translations/es_AR.ts Updating 'translations/es_AR.ts'... Found 0 source text(s) (0 new and 0 already existing)

I think you are missing the "." before the -ts option:
```
$ cd src/GUI.Qt/
$ lupdate-qt4 -source-language en_US -target-language es_AR . -ts translations/es_AR.ts
```

elKaZe, 2016-09-16:
was this :dancer: 

btw #137

dkopecek, 2016-12-30:
Fixed in master. The es_AR translation is now compiled in and shipped.

closed 2016-12-30 by dkopecek