USBGuard/usbguard #135: usbguard has problem with jabra device

bpereto, 2016-09-15:
```usbguard generate-policy``` makes my jabra bluetooth headset (via usb dongle) disapear. I wanted to let you know. Probably, there is work on the alsa site?

I use fedora (usbguard-0.5.10-1.fc24.x86_64)

```
Sep 15 18:36:19 mymachine pulseaudio[1869]: [alsa-sink-USB Audio] alsa-util.c: Got POLLNVAL from ALSA
Sep 15 18:36:19 mymachine pulseaudio[1869]: [alsa-sink-USB Audio] alsa-util.c: Could not recover from POLLERR|POLLNVAL|POLLHUP with snd_pcm_prepare(): No such device
Sep 15 18:36:19 mymachine /usr/libexec/gdm-x-session[1742]: (II) config/udev: removing device Jabra LINK 360
Sep 15 18:36:19 mymachine /usr/libexec/gdm-x-session[1742]: (II) evdev: Jabra LINK 360: Close
Sep 15 18:36:19 mymachine /usr/libexec/gdm-x-session[1742]: (II) UnloadModule: "evdev"
Sep 15 18:36:19 mymachine /usr/libexec/gdm-x-session[1742]: (II) systemd-logind: releasing fd for 13:69
Sep 15 18:36:19 mymachine kernel: usb 3-3.3: USB disconnect, device number 39
```

dkopecek, 2016-09-15:
Hello and thanks for the report! I think this issue should be resolved since 0.5.14. There's an update to usbguard 0.6.0 available in the Fedora updates-testing repository.

Alternatively, you could enable the [usbguard Copr repository](https://copr.fedorainfracloud.org/coprs/mildew/usbguard/) and install an updated package from there -- there's the latest development version available.

Please note that there are some backwards incompatible changes since 0.5.10, so you might have to re-generate your policy after the update. See the [Release page](https://github.com/dkopecek/usbguard/releases) for details about the changes.

dkopecek, 2016-09-17:
I've released usbguard-0.6.1. It's available in Fedora updates-testing repo, please test and let me know whether it fixes the issue with your device. To update from the testing repo, run:
```
$ sudo dnf update --enablerepo=updates-testing usbguard
```

dkopecek, 2016-10-26:
I consider this as fixed in the latest release. Please reopen if that's not the case.

closed 2016-10-26 by dkopecek