USBGuard/usbguard #138: QT5 applet compilation error

zaolin, 2016-09-18:
Hey,

I get an error compiling the qt5 applet:

 src/GUI.Qt/DeviceDialog.ui.h:13:25: fatal error: QtGui/QAction: No such file or directory

For QT5 the QAction header is located in the /usr/include/qt5/QtWidgets/QAction instead of QtGui.

BG Zaolin

dkopecek, 2016-09-19:
Hi, are you using the latest sources? I can't find a reference to QAction from any of the DeviceDialog related files.

zaolin, 2016-09-19:
Hey, I am using the HEAD (master). It seems that the file is auto generated by the wrong qt user interface compiler (4.8.6). I have selected --with-gui-qt=qt5 to build the qt5 applet

dkopecek, 2016-09-19:
Could you please attach the output of ./configure? It seems to be an issue with autodetection of the Qt tools to use during the compilation.

zaolin, 2016-09-19:
[config.txt](https://github.com/dkopecek/usbguard/files/479852/config.txt)

dkopecek, 2016-09-21:
You'll have to install qt5 linguist tools on your system. I think you have qt4 linguist there instead and the autodetection is picking them up via the generic command name (e.g. lrelease instead of lrelease-qt4).

zaolin, 2016-09-22:
For Gentoo QT_SELECT needs to be exported and set to the matching qt version in order to work.

dkopecek, 2016-09-26:
Thanks! I'll close this ticket as the referenced PR was merged.

closed 2016-09-26 by dkopecek