USBGuard/usbguard #145: USB Guard on Ubuntu 16.04

MatejKovacic, 2016-12-12:
Hi,

I installed USB Guard on Ubuntu 16.04 system:

git clone https://github.com/dkopecek/usbguard
cd usbguard/
./autogen.sh

sudo apt install libgcrypt11-dev libsodium-dev protobuf-compiler libprotobuf-dev libdbus-1-dev libdbus-glib-1-dev libpolkit-gobject-1-dev

./configure --prefix=/usr --sysconfdir=/etc --with-bundled-catch --with-bundled-pegtl
make
sudo make install
sudo ldconfig

However, when I try to start it, it says:
sudo systemctl start usbguard.service

Failed to start usbguard.service: Unit usbguard.service not found.

Also, there is no GUI (or at least I am unable to find it)... Any help will be much appreciated.

MatejKovacic, 2016-12-12:
Hi,

I already figured out how to do it. Here is a complete "receipt", you can use it on Wiki:
```
git clone https://github.com/dkopecek/usbguard
sudo apt install libgcrypt11-dev libsodium-dev protobuf-compiler libprotobuf-dev libdbus-1-dev libdbus-glib-1-dev libpolkit-gobject-1-dev
cd usbguard/
./autogen.sh
./configure --prefix=/usr --sysconfdir=/etc --with-bundled-catch --with-bundled-pegtl --with-gui-qt=qt4 --enable-systemd
make
sudo make install
sudo ldconfig

sudo usbguard generate-policy > rules.conf
sudo install -m 0600 -o root -g root rules.conf /etc/usbguard/rules.conf

sudo systemctl restart usbguard

usbguard-applet-qt
```

Works like a charm!

dkopecek, 2016-12-15:
Hi, thanks for the receipt! I have one more enhancement. Don't not install both libsodium-dev and libgcrypt11-dev. You need only one of those. If you want gcrypt for example, use the --with-crypto-library=gcrypt and install just libgcrypt11-dev.

fauno, 2017-04-20:
also missing `libqt4-dev libseccomp-dev xsltproc` :)

subversive-owl, 2017-06-18:
Also, `sudo apt install libtool`

fcsonline, 2017-06-23:
The `usbguard-applet-qt` was not working for me in `HEAD`. I just checked the last stable tag out before build it:

`git checkout usbguard-0.7.0`

i8x4t, 2017-08-09:
I get the following error trying to run autogen.sh on Ubuntu 16.04 `configure.ac:97: error: possibly undefined macro: AC_MSG_ERROR
      If this token and others are legitimate, please use m4_pattern_allow.
      See the Autoconf documentation.
configure.ac:155: error: possibly undefined macro: AC_DEFINE
configure.ac:166: error: possibly undefined macro: AC_MSG_FAILURE
configure.ac:236: error: possibly undefined macro: AC_MSG_NOTICE
configure.ac:252: error: possibly undefined macro: AC_CHECK_PROGS
autoreconf: /usr/bin/autoconf failed with exit status: 1
`

closed 2019-03-11 by dkopecek