USBGuard/usbguard #146: usbguard-applet-qt does not require sudo privileges

MatejKovacic, 2016-12-12:
Hi,
in usbguard-daemon.conf file you can set who is allowed to set rules:
IPCAllowedUsers=root
IPCAllowedGroups=wheel

The first problem, which is by my opinion also seriuos security problem, is that if group does not exists on a system, anyone is able to set the rules via usbguard-applet-qt. For instance, on Debian-like systems there is no wheel group by default, so by default setting any user can allow USB devices.

On the other side, if I set 
IPCAllowedUsers=root
IPCAllowedGroups=sudo

and I am a member of sudo, applet does not ask me for a sudo password. It would make sense if applet would ask you for your sudo password in any case.
If not, anyone can set a new rule while you are not around your computer for a few moments (and you forgot to invoke screen password).

Also, if user does not have a privilege to change rules, it would be great it USB Guard would display a notification - "This USB device has been blocked, please contact your system administrator to enable it." (or smth like that).

What do you thing?

grawity, 2017-07-15:
If the applet used polkit, then password prompts would automatically become available (depending on whether the sysadmin changes polkit rule to `allow` or `auth_self` or `auth_admin`.)

BenjaminSchubert, 2017-07-16:
@grawity there is work in progress on my branch to implement this :)

MatejKovacic, 2017-12-13:
How is your branch developing?

BenjaminSchubert, 2017-12-14:
I didn't have time to work on this at all since July. I should have more time during the winter break for this. If someone wants to help, I can take some time to clean my local code and push it so we can collaborate.

closed 2019-03-11 by dkopecek

miksuh, 2020-05-23:
Usbguard applet reallyy should be made more secure if that is not yet implemented. It really should ask password when you try to allow USB -device.

genodeftest, 2020-05-24:
> Usbguard applet reallyy should be made more secure if that is not yet implemented. It really should ask password when you try to allow USB -device.

The applet has been removed from this repository. Anyway, `sudo` is dangerous. I recommend not using it. Also, I don't see the benefit of using `sudo` here.

miksuh, 2020-05-24:
> > Usbguard applet reallyy should be made more secure if that is not yet implemented. It really should ask password when you try to allow USB -device.
> 
> The applet has been removed from this repository. Anyway, `sudo` is dangerous. I recommend not using it. Also, I don't see the benefit of using `sudoOk I did not know that the applet was removed I did not even mention sudo, all I said was that usbguard applet should ask a password. In my opinion it was a big security risk that the usbguard applet allowed user to accept USB devices whitout asking a password. I did not even think about sudo, more something like policykit etc. As a Debian user I actually am not so big fan of sudo, atleast not fan of the way Ubuntu is using sudo as a replacement of root.  .

genodeftest, 2020-05-24:
@miksuh: The syntax of your comment seems to break with GitHub's Markdown interpreter.

> I did not even mention sudo

That's what this issue is about, see the title.

> In my opinion it was a big security risk that the usbguard applet allowed user to accept USB devices whitout asking a password

The solution is not adding the user to the group which is permitted to use usbguard. See documentation.

miksuh, 2020-05-24:
e

> @miksuh: The syntax of your comment seems to break with GitHub's Markdown interpreter.
>
>
> That's what this issue is about, see the title.
>
>
> The solution is not adding the user to the group which is [...]

Again  I DID NOT COMMENT THE SUDO PART. In the earlier comment in this thread policykit and password was menitoned and that was more it what I commented. I did comment the fact that usbguard applet was insecure when it did not ask a password. I did not comment any way it how the password requirement should be implemented. But someone else already mentioned policykit in the earlier commen. I actually was thing something like policykit.

I am sorry to say this but in my opinion it was somekind of brain fart in the first place that you developed usbguard to block unknown usb devices and then you developed usbguard applet which allowed anyone to accept any device whitout even asking a password. This thread was the only and firs I found when I tried to find out if that problem with the usbguard applet is fixed. Obviously it never was fixed then. And that is a brain fart and it is good if thatt applet is gone now.t

genodeftest, 2020-05-24:
@miksuh : Please use fair and appropriate words. No need to insult people's actions.

Please try to understand the security model/concept before criticizing it. In short, the concept is protecting against devices plugged into a computer without the user knowing about (drive-by attacks, keyloggers). Also, it intends to help users against malicious devices aka *BadUSB* (e.g. an USB thumb drive with a manipulated firmware which identifies as USB keyboard to the computer, opens a shell and executes a command). USBguard is not about preventing users from using USB (specific) devices at all.