USBGuard/usbguard #148: Linux root hub USB descriptor is not stable across kernel versions

endomandi, 2016-12-17:
I include the output of `lsusb -v` for one of the root hubs on my machine. Note that iManufacturer includes the kernel version; this is problematic when using the parent hash functionality.

The hash is blindly of the entire USB descriptor as binary data, so I'm not sure what approach to take to ameliorate this.

```
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.00
  bDeviceClass            9 Hub
  bDeviceSubClass         0 Unused
  bDeviceProtocol         3 
  bMaxPacketSize0         9
  idVendor           0x1d6b Linux Foundation
  idProduct          0x0003 3.0 root hub
  bcdDevice            4.08
  iManufacturer           3 Linux 4.8.0-1-amd64 xhci-hcd
  iProduct                2 xHCI Host Controller
  iSerial                 1 0000:00:14.0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           31
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower                0mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0 Unused
      bInterfaceProtocol      0 Full speed (or root) hub
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0004  1x 4 bytes
        bInterval              12
        bMaxBurst               0
Hub Descriptor:
  bLength              12
  bDescriptorType      42
  nNbrPorts             4
  wHubCharacteristic 0x000a
    No power switching (usb 1.0)
    Per-port overcurrent protection
  bPwrOn2PwrGood       10 * 2 milli seconds
  bHubContrCurrent      0 milli Ampere
  bHubDecLat          0.0 micro seconds
  wHubDelay             0 nano seconds
  DeviceRemovable    0x00
 Hub Port Status:
   Port 1: 0000.02a0 5Gbps power Rx.Detect
   Port 2: 0000.02a0 5Gbps power Rx.Detect
   Port 3: 0000.02a0 5Gbps power Rx.Detect
   Port 4: 0000.02a0 5Gbps power Rx.Detect
Binary Object Store Descriptor:
  bLength                 5
  bDescriptorType        15
  wTotalLength           15
  bNumDeviceCaps          1
  SuperSpeed USB Device Capability:
    bLength                10
    bDescriptorType        16
    bDevCapabilityType      3
    bmAttributes         0x00
    wSpeedsSupported   0x0008
      Device can operate at SuperSpeed (5Gbps)
    bFunctionalitySupport   3
      Lowest fully-functional device speed is SuperSpeed (5Gbps)
    bU1DevExitLat          10 micro seconds
    bU2DevExitLat         512 micro seconds
Device Status:     0x0001
  Self Powered
```

endomandi, 2016-12-28:
Ah, guilty of jumping slightly to conclusions there. It's probably bcdDevice causing the problems, as it's only the string indexed by iManufacturer that changes, not iManufacturer itself.

dkopecek, 2016-12-31:
Hi and thanks for the report! Looks like the Linux root hub will be a problematic device that will require exceptions. So far I don't have a solution nor a workaround for this.

dkopecek, 2016-12-31:
For reference: https://github.com/torvalds/linux/blob/e6445f52d9c8b0e6557a45fa7d0e8e088d430a8c/drivers/usb/core/hcd.c#L165

Possible solutions:
 1. Don't include the bcdDevice value in the device hash (i.e. make an in-memory copy of the descriptor, reset bcdDevice to some constant value and compute the hash using the modified descriptor).
 2. Don't include the bcdDevice value just for the Linux root hub (based on vendor_id, product_id) and instead check that it's correct (set to the expected KERNEL_VER, KERNEL_REL).

commit dbfa9b8f (2017-01-11): Reset Linux root hub bcdDevice value before updating device hash

The Linux root hub USB device uses KERNEL_VER and KERNEL_REL
as the bcdDevice value. Since we compute hash over the descriptor
data, the changing bcdDevice causes the hash to change with every
new kernel release. This commit fixes this by reseting the bcdDevice
value to 0 before updating the device hash value.

Reference:
https://github.com/torvalds/linux/blob/e6445f52d9c8b0e6557a45fa7d0e8e088d430a8c/drivers/usb/core/hcd.c#L165

Resolves: #148
closed 2017-01-11 by dkopecek