USBGuard/usbguard #151: Usbguard doesn't apply to early boot stages

genodeftest, 2017-01-02:
In early boot, i.e. before mounting the root filesystem, usbguard doesn't work. To fix this, it probably needs to be saved into initramfs and loaded pretty soon in the boot process.

I ran into this issue because of another bug where usbguard would disable all USB devices but allow USB keyboard to insert the cryptsetup HDD password. After that, the keyboard goes dark.

Caveat:
Fixing this bug might be hard because you'll need to make sure that the initramfs is regenerated every time USB device rules change.

genodeftest, 2017-08-19:
Workaround: Remove all unused kernel modules from your initramfs, e.g. for removable USB storage drivers or vfat.

mroethke, 2017-12-30:
Another option is setting `usbcore.authorized_default=0` on the kernel command line.

genodeftest, 2018-01-03:
> Another option is setting `usbcore.authorized_default=0` on the kernel command line.

This is a nice workaround, but it only works if you don't need USB devices to unlock your disk. Having an USB keyboard, adding this option to kernel command line renders me unable to enter my LUKS password.

It seems like having usbguard plus its config files in initramfs is required in combination to what you suggested above.

mrwiora, 2025-11-29:
@mroethke most probably you meant a TPM right?

to all - for my notebook installation the "usbcore.authorized_default=0" cmdline parameter works perfectly, as the keyboard is connected via i8042 as as PS/2 device (this is a very latest HP 830 G11 Notebook).

mroethke, 2025-12-07:
On 2025-11-29 13:48:18, mrwiora wrote:
> @mroethke most probably you meant a TPM right?

No, I did not.