USBGuard/usbguard #152: [Qt Applet] Don't jump the accept/block dialog around

genodeftest, 2017-01-02:
Currently, when attaching a new USB device, the user is prompted by a dialog to accept or block the new device. This device is being displayed at different screen positions on successive tries. I think I read that this is a "security" feature, but it isn't: In case the attacker has access to the mouse or cursor, he will also have access to the physical or virtual keyboard, rendering this "defense" mechanism unusable.

genodeftest, 2017-05-02:
This is now configurable, as in "Randomize position of the window".

I'll leave this issue open because I still think this feature should be removed. It is __not__ a security feature!

BenjaminSchubert, 2017-07-04:
I agree with @genodeftest this brings a false sense of security.

@dkopecek should I remove this in #178 ?

dkopecek, 2017-07-04:
> @dkopecek should I remove this in #178 ?

Yes, thanks!

vk5tu, 2017-11-13:
It's insecure in a common use-case. My authorise dialog was displayed in front of a few hundred people because it chose to display on the projector screen rather than then laptop screen.

genodeftest, 2017-11-13:
@vk5tu :
> It's insecure in a common use-case. My authorise dialog was displayed in front of a few hundred people because it chose to display on the projector screen rather than then laptop screen.

This is a separate issue and I guess you should file a separate bug report for that.

closed 2019-03-11 by dkopecek