USBGuard/usbguard #153: Update to 0.6.3 breaks usbguard on boot

genodeftest, 2017-01-02:
It looks like every 0.6.3 release from copr is broken. 0.6.2-1.fc25.x86_64 from Fedora repos works fine.

What happens:
On boot, USB devices work fine in initramfs, e.g. to insert the cryptsetup HDD encryption password. After the kernel switches to rootfs on disk, all USB devices are disabled rendering the system useless. A later (successful) reboot reveals that this is printed to syslog:
```
$ journalctl -b -4 | grep usbguard
Jan 02 18:28:46 hostname audit[1]: SERVICE_START pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=usbguard comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
Jan 02 18:28:46 hostname usbguard-daemon[807]: [1483378126.942] (E) Device insert exception: UEventDeviceManager: /sys/devices/pci0000:00/0000:00:1d.0/usb2: unknown syspath
Jan 02 18:28:46 hostname usbguard-daemon[807]: [1483378126.942] (W) UEventDeviceManager: /sys/devices/pci0000:00/0000:00:1d.0/usb2: unknown syspath
Jan 02 18:28:46 hostname usbguard-daemon[807]: [1483378126.942] (W) Rejecting device at syspath=/sys/devices/pci0000:00/0000:00:1d.0/usb2/2-1
Jan 02 18:28:47 hostname usbguard-daemon[807]: [1483378127.063] (E) Device insert exception: UEventDeviceManager: /sys/devices/pci0000:00/0000:00:1a.0/usb1/1-1: unknown syspath
Jan 02 18:28:47 hostname usbguard-daemon[807]: [1483378127.063] (W) UEventDeviceManager: /sys/devices/pci0000:00/0000:00:1a.0/usb1/1-1: unknown syspath
Jan 02 18:28:47 hostname usbguard-daemon[807]: [1483378127.063] (W) Rejecting device at syspath=/sys/devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.3
Jan 02 18:28:47 hostname usbguard-daemon[807]: [1483378127.072] (E) Device insert exception: UEventDeviceManager: /sys/devices/pci0000:00/0000:00:1a.0/usb1/1-1: unknown syspath
Jan 02 18:28:47 hostname usbguard-daemon[807]: [1483378127.072] (W) UEventDeviceManager: /sys/devices/pci0000:00/0000:00:1a.0/usb1/1-1: unknown syspath
Jan 02 18:28:47 hostname usbguard-daemon[807]: [1483378127.072] (W) Rejecting device at syspath=/sys/devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.1
Jan 02 18:28:47 hostname usbguard-daemon[807]: [1483378127.101] (W) USB Device Exception: SysFSDevice: (_sysfs_dirfd = open(_sysfs_path.c_str(), O_PATH|O_DIRECTORY)) < 0: No such file or directory
Jan 02 18:28:47 hostname usbguard-daemon[807]: [1483378127.101] (W) SysFSDevice: (_sysfs_dirfd = open(_sysfs_path.c_str(), O_PATH|O_DIRECTORY)) < 0: No such file or directory
Jan 02 18:28:47 hostname usbguard-daemon[807]: [1483378127.105] (W) USB Device Exception: SysFSDevice: (_sysfs_dirfd = open(_sysfs_path.c_str(), O_PATH|O_DIRECTORY)) < 0: No such file or directory
Jan 02 18:28:47 hostname usbguard-daemon[807]: [1483378127.105] (W) SysFSDevice: (_sysfs_dirfd = open(_sysfs_path.c_str(), O_PATH|O_DIRECTORY)) < 0: No such file or directory
```

Workaround:
* downgrade to 0.6.2 which used to work fine. Note that the config changed.
* add `systemd.mask=usbguard.service` to your kernel command line

I'm on Kernel 4.8.15-300.fc25.x86_64.

dkopecek, 2017-01-02:
@genodeftest thanks for testing the bleeding edge copr builds :D I'm aware of this problem and I'm working on a fix.

Are you using the copr repo on a production system? I started to use it also for building the testing releases (the ones with -0.1.<timestamp> release suffix) which will probably often break things.

genodeftest, 2017-01-02:
> Are you using the copr repo on a production system? I started to use it also for building the testing releases (the ones with -0.1. release suffix) which will probably often break things.

Kind of, yes. I'll be more careful now I guess ;)

genodeftest, 2017-01-02:
And the `usbguard generate-policy` is broken too.

dkopecek, 2017-01-16:
@genodeftest the latest build should work fine (incl. generate-policy), would you have some time to test it?

genodeftest, 2017-01-16:
Rebooting works fine with the `usbguard-0.6.3-0.1.1484332211.fc25.x86_64` build from mildew/usbguard repo from Friday, 13th of January. Thank you!

Now I'm only getting one warning:
`(W) udev backend is OBSOLETE. Falling back to new default: uevent`
but that's expected since 0.6.3 introduces uevent and I still use the old config and policy files.

And the update tried to clean my `rules.conf` file, but the .rpmnew feature prevented this.

genodeftest, 2017-01-16:
I got rid of the warning by updating the `usbguard-daemon.conf` file with the changes suggested by the RPM file/update.

I can also confirm that `usbguard generate-policy` works fine again.

I still left this report open because of trying to delete my rules.conf file.

genodeftest, 2017-01-17:
> I still left this report open because of trying to delete my rules.conf file.

This is also fixed now in `usbguard-0.6.3-0.1.1484644459.fc25.x86_64`. Thank you!

closed 2017-01-17 by genodeftest