USBGuard/usbguard #154: From the repo, packages pushed at the same time for both Fedora and CentOS

phocean, 2017-01-03:
Hello,

Since the last Copr repo update, there is something wrong with Fedora / CentOS packages.

I now get warnings about broken dependancies all the time:

```
%  sudo dnf update            
[...]
Paquets ignorés suite à des dépendances cassées:
 usbguard                x86_64      0.6.3-0.1.1483289356.el7.centos       mildew-usbguard      347 k
 usbguard-applet-qt      x86_64      0.6.3-0.1.1483289356.el7.centos       mildew-usbguard       67 k
[...]
```
As I am running Fedora, I should not be offered CentOS packages. And naturally, dependancies don't match.

Note that the Fedora packages are offered and installed at the same time:

```
%  dnf list installed | grep usbguard
usbguard.x86_64                      0.6.2-1.fc25               @fedora         
usbguard-applet-qt.x86_64            0.6.2-1.fc25               @fedora
```

dkopecek, 2017-01-07:
Hello, try removing the package and installing again. I'm not sure why it tries to install the .el7 package though :\

Anyway, I would recommend not to install the packages with -0.1.<timestamp> release suffix. Those are only testing builds. I'll create a special repository for them once I release 0.6.3.

phocean, 2017-01-08:
Reinstalling always pick up the right Fedora packages, so not really a problem except this warning.

dkopecek, 2017-05-10:
Closing. I will build the testing releases in a separate Copr repository from now on:

https://copr.fedorainfracloud.org/coprs/mildew/usbguard-testing/

closed 2017-05-10 by dkopecek