USBGuard/usbguard #156: Blocks usb_modeswitch operation

intrigeri, 2017-01-21:
Hi!

I initially thought I got my rules wrong, but even with `ImplicitPolicyTarget=allow` and an empty ruleset, usbguard 0.6.2 prevents usb_modeswitch from doing its job: my 3G USB stick remains in its initial mode (as plugged) and is never switched to modem mode.

Am I doing something wrong?

Cheers!

dkopecek, 2017-01-23:
Hi! Could you please run the daemon with debugging enabled and attach the logs? You can enable debugging by adding the -d option to command-line arguments when starting the daemon. Something like:
```
# usbguard-daemon -d -k -c /etc/usbguard/usbguard-daemon.conf
```

intrigeri, 2017-02-07:
> Could you please run the daemon with debugging enabled and attach the logs?

Sure. I've redacted the logs a bit though, because I'd rather not leak
all my devices serial numbers etc. here.

With usbguard running:

<pre>
Feb 07 13:55:08 localhost usbguard-daemon[396]: [1486472108.711] (D) LinuxDeviceManager.cpp@328/thread: UDev read event.
Feb 07 13:55:08 localhost usbguard-daemon[396]: [1486472108.711] (T) LinuxDeviceManager.cpp@480/processDeviceRemoval: dev=0x7f120c02a2f0
Feb 07 13:55:08 localhost usbguard-daemon[396]: [1486472108.711] (T) DeviceManagerPrivate.cpp@58/removeDevice: entry: id=15
Feb 07 13:55:08 localhost usbguard-daemon[396]: [1486472108.711] (T) DeviceManagerPrivate.cpp@66/removeDevice: return: device_ptr=0x7f120c01dfc0
Feb 07 13:55:08 localhost usbguard-daemon[396]: [1486472108.711] (T) DeviceManagerPrivate.cpp@96/DeviceEvent: event=Removedevice_ptr=0x7f120c01dfc0
Feb 07 13:55:08 localhost usbguard-daemon[396]: [1486472108.711] (T) Daemon.cpp@372/dmHookDeviceEvent: event=Remove device_ptr=0x7f120c01dfc0
Feb 07 13:55:08 localhost usbguard-daemon[396]: [1486472108.711] (T) DevicePrivate.cpp@69/getDeviceRule: entry:  with_port=1 with_parent_hash=1 match_rule=0
Feb 07 13:55:08 localhost usbguard-daemon[396]: [1486472108.711] (T) DeviceManagerPrivate.cpp@84/getDevice: id=13
Feb 07 13:55:08 localhost usbguard-daemon[396]: [1486472108.711] (T) DevicePrivate.cpp@110/getDeviceRule: return: device_rule=allow id VENDOR_ID:PRODUCT_ID serial "SERIAL_DELETED" name "REDACTED Modem" hash "koHgyPbLUkSoQUbzt6jSMpwxldSkW1TtWU1IGShQYTI=" parent-hash "Miigb8mx72Z0q6L+YMai0mDZSlYC8qiSMctoUjByF2o=" via-port "1-4" with-interface 08:06:50
Feb 07 13:55:16 localhost kernel: usb 1-4: new high-speed USB device number 27 using xhci_hcd
Feb 07 13:55:16 localhost kernel: usb 1-4: New USB device found, idVendor=2357, idProduct=f000
Feb 07 13:55:16 localhost kernel: usb 1-4: New USB device strings: Mfr=3, Product=2, SerialNumber=4
Feb 07 13:55:16 localhost kernel: usb 1-4: Product: REDACTED Modem
Feb 07 13:55:16 localhost kernel: usb 1-4: Manufacturer: REDACTED, Incorporated
Feb 07 13:55:16 localhost kernel: usb 1-4: SerialNumber: SERIAL_DELETED
Feb 07 13:55:16 localhost kernel: usb 1-4: Device is not authorized for usage
Feb 07 13:55:16 localhost systemd[1]: Starting USB_ModeSwitch_1-4...
Feb 07 13:55:16 localhost usbguard-daemon[396]: [1486472116.504] (D) LinuxDeviceManager.cpp@328/thread: UDev read event.
Feb 07 13:55:16 localhost usbguard-daemon[396]: [1486472116.504] (T) DeviceManagerPrivate.cpp@48/insertDevice: device_ptr=0x7f120c016080
Feb 07 13:55:16 localhost usbguard-daemon[396]: [1486472116.504] (D) DeviceManagerPrivate.cpp@51/insertDevice: id=21
Feb 07 13:55:16 localhost usbguard-daemon[396]: [1486472116.504] (T) DeviceManagerPrivate.cpp@96/DeviceEvent: event=Insertdevice_ptr=0x7f120c016080
Feb 07 13:55:16 localhost usbguard-daemon[396]: [1486472116.504] (T) Daemon.cpp@372/dmHookDeviceEvent: event=Insert device_ptr=0x7f120c016080
Feb 07 13:55:16 localhost usbguard-daemon[396]: [1486472116.504] (T) DevicePrivate.cpp@69/getDeviceRule: entry:  with_port=1 with_parent_hash=1 match_rule=0
Feb 07 13:55:16 localhost usbguard-daemon[396]: [1486472116.504] (T) DeviceManagerPrivate.cpp@84/getDevice: id=13
Feb 07 13:55:16 localhost usbguard-daemon[396]: [1486472116.504] (T) DevicePrivate.cpp@110/getDeviceRule: return: device_rule=block id VENDOR_ID:PRODUCT_ID serial "SERIAL_DELETED" name "REDACTED Modem" hash "koHgyPbLUkSoQUbzt6jSMpwxldSkW1TtWU1IGShQYTI=" parent-hash "Miigb8mx72Z0q6L+YMai0mDZSlYC8qiSMctoUjByF2o=" via-port "1-4" with-interface 08:06:50
Feb 07 13:55:16 localhost usbguard-daemon[396]: [1486472116.504] (T) Daemon.cpp@442/getDevicePolicyRule: device_ptr=0x7f120c016080
Feb 07 13:55:16 localhost usbguard-daemon[396]: [1486472116.504] (T) DevicePrivate.cpp@69/getDeviceRule: entry:  with_port=1 with_parent_hash=1 match_rule=1
Feb 07 13:55:16 localhost usbguard-daemon[396]: [1486472116.504] (T) DeviceManagerPrivate.cpp@84/getDevice: id=13
Feb 07 13:55:16 localhost usbguard-daemon[396]: [1486472116.504] (T) DevicePrivate.cpp@110/getDeviceRule: return: device_rule=match id VENDOR_ID:PRODUCT_ID serial "SERIAL_DELETED" name "REDACTED Modem" hash "koHgyPbLUkSoQUbzt6jSMpwxldSkW1TtWU1IGShQYTI=" parent-hash "Miigb8mx72Z0q6L+YMai0mDZSlYC8qiSMctoUjByF2o=" via-port "1-4" with-interface 08:06:50
Feb 07 13:55:16 localhost usbguard-daemon[396]: [1486472116.504] (T) RulePrivate.cpp@132/appliesToWithConditions: entry: rhs=match id VENDOR_ID:PRODUCT_ID serial "SERIAL_DELETED" name "REDACTED Modem" hash "koHgyPbLUkSoQUbzt6jSMpwxldSkW1TtWU1IGShQYTI=" parent-hash "Miigb8mx72Z0q6L+YMai0mDZSlYC8qiSMctoUjByF2o=" via-port "1-4" with-interface 08:06:50 with_updates=1
Feb 07 13:55:16 localhost usbguard-daemon[396]: [1486472116.504] (T) RulePrivate.cpp@105/appliesTo: entry: rhs=match id VENDOR_ID:PRODUCT_ID serial "SERIAL_DELETED" name "REDACTED Modem" hash "koHgyPbLUkSoQUbzt6jSMpwxldSkW1TtWU1IGShQYTI=" parent-hash "Miigb8mx72Z0q6L+YMai0mDZSlYC8qiSMctoUjByF2o=" via-port "1-4" with-interface 08:06:50 parent_insensitive=0
Feb 07 13:55:16 localhost usbguard-daemon[396]: [1486472116.504] (T) RulePrivate.cpp@179/appliesTo: entry: source=id 03f0:5607 target=id VENDOR_ID:PRODUCT_ID
Feb 07 13:55:16 localhost usbguard-daemon[396]: [1486472116.504] (D) RulePrivate.cpp@191/appliesTo: set_operator=equals
Feb 07 13:55:16 localhost usbguard-daemon[396]: [1486472116.504] (T) RulePrivate.cpp@214/appliesTo: return: applies=0
[... cut ...]
Feb 07 14:05:14 localhost kernel: sr 2:0:0:0: [sr0] scsi-1 drive
Feb 07 14:05:14 localhost kernel: sr 2:0:0:0: Attached scsi CD-ROM sr0
Feb 07 14:05:14 localhost kernel: sr 2:0:0:0: Attached scsi generic sg1 type 5
Feb 07 14:05:14 localhost kernel: sd 2:0:0:1: Attached scsi generic sg2 type 0
Feb 07 14:05:14 localhost kernel: sd 2:0:0:1: [sdb] Attached SCSI removable disk
Feb 07 14:05:14 localhost systemd[1]: Started USB_ModeSwitch_1-4.
[...]
Feb 07 14:05:45 localhost ModemManager[1194]: <info>  Creating modem with plugin 'Generic' and '3' ports
</pre>

grawity, 2017-07-15:
I'm guessing this is not an USBGuard bug, but a result of how the usb_modeswitch udev rules are written &ndash; they trigger as soon as the `usb_device` appears and blindly assume that all child `usb_interface`:s will be ready immediately.

But in your case, usb_modeswitch is unable to do anything and exits. When USBGuard finally authorizes the device, the child `usb_interface` shows up but there's no udev rule to react on it.

Your `(/usr)/lib/udev/rules.d/40-usb_modeswitch.rules` will have this line:

    ATTR{idVendor}=="2357", ATTR{idProduct}=="f000", RUN+="usb_modeswitch '/%k'"

Try adding a `/etc/udev/rules.d/41-modeswitch-harder.rules` with:

    ATTR{idVendor}=="2357", ATTR{idProduct}=="f000", ENV{modeswitch_dev}="%k"

    ENV{DEVTYPE}=="usb_interface", IMPORT{parent}="modeswitch_dev"

    ACTION=="add", ENV{modeswitch_dev}=="?*", RUN+="usb_modeswitch '/%E{modeswitch_dev}'"

Run `udevadm control --reload` after editing, then reconnect the device.

_(Also, no thanks for trying to redact the VID&PID from the log.)_

dkopecek, 2018-01-22:
@grawity Hi and thanks for helping with this! Do you think it would be a good idea for usbguard to provide (or even install) such a udev snippet to fix this situation?

grawity, 2018-01-22:
No, if it calls usb_modeswitch then it should remain part of usb_modeswitch. (You could say the existing snippet installed by usb_modeswitch is buggy regardless of usbguard.) The above snippet is just meant as a local workaround.

grawity, 2018-01-22:
No, if it calls usb_modeswitch then it should remain part of usb_modeswitch. (You could say the existing rules installed by usb_modeswitch are buggy regardless of usbguard.)

The above snippet is meant as a local workaround, and as an example of how usb_modeswitch _should_ fix its rules.

muelli, 2019-01-29:
has this been reported to the usb modeswitch folks yet?

intrigeri, 2019-03-30:
> has this been reported to the usb modeswitch folks yet? 

@muelli, at least I did not. I haven't access to the affected hardware anymore.