USBGuard/usbguard #157: Please allow hiding the systray icon while keeping notifications functionality

intrigeri, 2017-01-21:
Hi!

background: we're considering adding usbguard to Tails, so I've been evaluating it, mostly from a UX point of view.

I understand the desire to provide an easy way to fire up the control GUI, but I have a problem with the systray icon: everyone wants to add their own, and I don't need another one. What I need is:

* MUST: be notified when a device is allowed/blocked/whatever -- we have this already, awesome! :)
* MUST: a way to start the control GUI, e.g. when one needs to fix a rule they've added already -- we have this already, but it doesn't need to clutter the screen 24/7; I have other ways to start hundreds of GUIs on my system, and quite frankly this one is not particularly special enough to warrant distracting my eyes; at least on Debian there's a `usbguard-applet-qt.desktop` that does the job, except it really starts the applet (that's already running if we want notifications), which I think points us to the root of the problem; see below
* MAY: let the user know whenever usbguard is not working or badly configured (e.g. when permissions prevent the control GUI from connecting to the backend daemon); I'm told this is currently done by the color of the systray icon; I suggest it could be done with one single notification warning on startup, and again is no sufficient reason to add yet another icon 24/7 on my screen.

So, I think that what I need is:

* some way to get notifications without adding one more icon 24/7 on my screen; a hack could be to allow hiding the systray icon, and then I get essentially what I need, without much work :)   We can refine later, if you want, how exactly this should be implemented, because of course it's kind of a hack (food for thought: a headless program, run in the desktop session, that uses the native desktop notification system, independently from the systray icon; I'll let those who want the icon design what exactly it should do).
* a control GUI, that can be started by a .desktop file, and is independent from notifications and from the systray icon interface; we have the GUI already, all that's needed is to make it standalone.
* whatever piece of code handles the notifications can warn the user, on startup, if it can't connect to the usbguard backend.

What do you think?

dkopecek, 2017-01-23:
Hello and thanks for the feedback! 

I'm all in for enhancing the GUI component of usbguard. Or creating a new and better GUI. The current one is basically a proof-of-concept and the UX is horrible. I like the changes you are proposing. See my questions inline:

> some way to get notifications without adding one more icon 24/7 on my screen;
> a hack could be to allow hiding the systray icon, and then I get essentially what I need,
> without much work :) We can [...]

Some questions here:
 1. What is the native desktop notification system?
 2. Any idea how to (re)implement the device authorization dialog? (the dialog window which asks you for a allow/block decision if there's no matching rule)

> a control GUI, that can be started by a .desktop file, and is independent from notifications
> and from the systray icon interface; we have the GUI already, all that's needed is to make
> it standalone.

Ok, this is easy to do.

> whatever piece of code handles the notifications can warn the user, on startup, if it can't
> connect to the usbguard backend.

+1

Just to wrap up. We should end up with:

1. Small **notification forwarder/bridge** (running in background) for:
   - notifying about the daemon not working as expected
   - notifying about device allow, block, reject and exception events

2. **Control GUI** (running only on-demand) for:
   - listing the authorization state of present devices
   - modifying the authorization state of present devices (if allowed by whatever IPC systems is in place)
   - listing the policy
   - modifying the policy (if allowed by whatever IPC systems is in place)

3. Something that asks the user for an authorization decision when there's no matching rule in the policy.
   - what? should that be part of the notification forwarder?

BenjaminSchubert, 2017-07-06:
@intrigeri I implemented this once I moved to DBus for notifications.

Could you please give it a look and tell me whether this seems good for you ? You can find the implementation at https://github.com/BenjaminSchubert/usbguard/tree/icon-removal. 

Would this behavior suit you better ? what would you change ?

intrigeri, 2017-07-06:
> @intrigeri I implemented this once I moved to DBus for notifications.

Cool, thanks for working on this! :)

> Could you please give it a look and tell me whether this seems good for you ? You can
> find the implementation at
> https://github.com/BenjaminSchubert/usbguard/tree/icon-removal.

> Would this behavior suit you better ? what would you change ?

Is it OK if I reply later this month, probably around July 15-20?

BenjaminSchubert, 2017-07-06:
> Is it OK if I reply later this month, probably around July 15-20?

Yes no worries, I have plenty of changes in the pipeline anyways :)

genodeftest, 2018-10-02:
*bump*, maybe related: #257

dkopecek, 2019-06-28:
Closing this as I've removed the applet from the main usbguard repository in https://github.com/USBGuard/usbguard/commit/bf3bb09edf491892617d35b55edec06f92d2743f. The plan is to simplify it down to a minimal required use case and maintain it in a separate repository.

closed 2019-06-28 by dkopecek

genodeftest, 2020-02-05:
If you are only interested in notifications, you may want to use https://github.com/Cropi/usbguard-notifier.