USBGuard/usbguard #158: Duplicate interfaces in set

cgzones, 2017-01-30:
Hi,
I am testing usbguard on my debian laptop at version 0.6.2+ds1-1.
For my webcam I am getting the following rule generated:
```
allow id 064e:9700 serial "HF1019-T838-SN03-Rev0101" name "USB2.0 UVC HD Webcam" hash "K1N9DW1XmzKwFXyOWC2jEs5F+2FIi/izW8T2q3q7B9I=" parent-hash "Miigb8mx72Z0q6L+YMai0mDZSlYC8qiSMctoUjByF2o=" with-interface { 0e:01:00 0e:02:00 0e:02:00 0e:02:00 0e:02:00 0e:02:00 0e:02:00 0e:02:00 0e:02:00 }
```
Why is the interface `0e:02:00` contained multiple times?

dkopecek, 2017-01-31:
Hi, thanks for trying out usbguard! USB webcams usually have an interface (of the same class:subclass:protocol) for each resolution they support (and other kinds of video parameters). The applications then choose a specific interface depending on the picture resolution they want to receive.

You can check that the device has so many interfaces using lsusb -v.

cgzones, 2017-01-31:
Thanks for clarification.
Would it make sense for rule generation to skip duplicate interfaces, so the result would look like:
```
allow id 064e:9700 serial "HF1019-T838-SN03-Rev0101" name "USB2.0 UVC HD Webcam" hash "K1N9DW1XmzKwFXyOWC2jEs5F+2FIi/izW8T2q3q7B9I=" parent-hash "Miigb8mx72Z0q6L+YMai0mDZSlYC8qiSMctoUjByF2o=" with-interface { 0e:01:00 0e:02:00 }
```

Btw, could you please document the value `keep` for the options `PresentControllerPolicy`  and `PresentControllerPolicy` a bit more?

Btw no 2, why does usbguard-daemon open its configuration file in write mode?
https://github.com/dkopecek/usbguard/blob/master/src/Library/ConfigFilePrivate.cpp#L45