USBGuard/usbguard #159: dbus complains about configuration

cgzones, 2017-02-03:
With usbguard version 0.6.2+ds1-1 on debian, dbus issues:
```
Feb  3 15:45:31 debainl dbus-daemon[639]: Unknown username "usbguard-dbus" in message bus configuration file
```
https://github.com/dkopecek/usbguard/blob/master/src/DBus/org.usbguard.conf#L13

Is this entry a leftover or is the usbguard-dbus daemon (in the future) supposed to run under this user?

hellyberry, 2017-02-11:
seen this also today

dkopecek, 2017-02-16:
Hi. Yes, the special user is supposed to be used in the future. I'll comment it out for now. Maybe it would be enough to just drop capabilities and stay as root because using a special user for the same purpose is harder w.r.t. distribution integration.

commit 65e3e542 (2017-03-05): Deactivate usbguard-dbus user configuration for D-Bus

Addressing:
```
dbus-daemon[]: Unknown username "usbguard-dbus" in message bus
configuration file
```

Resolves: #159
closed 2017-03-05 by dkopecek