USBGuard/usbguard #161: `Connection refused` error

indolering, 2017-02-13:
I'm just getting the following error message:

    service=usbguard: Connection refused
    [Mon Feb 13 14:26:48 2017] Connection failed: IPC connect:

What trouble shooting steps do I need to take?

dkopecek, 2017-02-16:
Hi, the "connection refused" error probably means that you don't have usbguard-daemon running. Could you check that? How did you install usbguard and what distribution are you using?

indolering, 2017-02-17:
Fedora, I installed it using DNF and ran `sudo systemctl start usbguard.service`....

dkopecek, 2017-02-17:
@indolering did you generate a policy using usbguard generate-policy? What is the output of systemctl status usbguard?

indolering, 2017-02-17:
> did you generate a policy using usbguard generate-policy

Ahh, now it's working!  Why wouldn't usbguard generate one by default on install?  I guess this is just an old project that doesn't get enough love?

dkopecek, 2017-03-05:
I don't think that generating the policy automatically is a good idea. Generating/Writing one is simply a required configuration step after installation.

closed 2017-03-05 by dkopecek

indolering, 2017-03-19:
> I don't think that generating the policy automatically is a good idea.

The program should at least offer to generate one instead of just failing.

pdolinic, 2022-02-22:
Getting the same message on Fedora 35 with SELinux enabled, and `usbguard generate-policy` and getting locked out after `systemctl start usbguard`. The keyboard has a USB-Hub though.

hartwork, 2022-02-22:
Hi @pdolinic this ticket is closed and there are a lot of open questions from your report, and I'm not a big fan creating fear when it may just turn out unjustified after a closer look. If you want to fix debug and fix this, please reach out via my e-mail and we can jump on a voice call about this in no time, and solve this. If that's not a good fit for you, I would ask to create a new ticket with enough information that it will be hard to ask a question that isn't already answered. Thank you, Sebastian

pdolinic, 2022-02-22:
Hello, I just managed to find the fix /details on the Arch-Wiki: https://wiki.archlinux.org/title/USBGuard

For some reason it was required to manually append the configuration on Fedora 35.

`usbguard generate-policy > /etc/usbguard/rules.conf`

Everything working now thanks!

hartwork, 2022-02-22:
Hi @pdolinic if I'm reading that the right way, you were running `usbguard generate-policy` (as root) earlier, but did not write/redirect its output to `/etc/usbguard/rules.conf`. That would explain the lockout. Cool that it's working for you now and thanks for the update :+1:

mYnDstrEAm, 2024-04-21:
Agree with indolering that there should be some prompt to generate the policy and set it up. 

Just trying to generate the policy and starting it would lock people off the computer. To prevent this here is the full instruction on Debian12 which should probably be in the readme and the user be guided through when first running it (or after install):

* `sudo apt-get install usbguard usbguard-notifier`
* not `sudo usbguard generate-policy` but `sudo su` and then `usbguard generate-policy > /etc/usbguard/rules.conf`
* `sudo usbguard add-user username`
* Ctrl+D then `sudo systemctl start usbguard.service` to start it (before doing that you could run `usbguard list-devices` to make sure it does list your keyboard)
* `usbguard-notifier` (if the notifier which would display a notification whenever a new USB device is plugged in is not running)

Now I'm wondering how to make it autostart since it doesn't do that after this setup but I'll make a new issue for that.