USBGuard/usbguard #162: usbguard list-devices -b : termination

cgzones, 2017-02-14:
Listing blocked devices when no are present causes usbguard to terminate:

```
root@raspberrypi:/root# usbguard list-devices   
4: allow id 1d6b:0002 serial "3f980000.usb" name "DWC OTG Controller" hash "EYx+GlFlkA3XIp4+eFRwuSFQM3DN0fehOlLVfI65aWw=" parent-hash "z+EE0r7/c8O/LqMIXf34DcLJgMjgneWBmFWXqrBWzsc=" via-port "usb1" with-interface 09:00:00
5: allow id 0424:9514 serial "" name "" hash "E0MWcMEgHt8SbS4HZnYsm7CORelcUwUjAtJKvvAP8yg=" parent-hash "EYx+GlFlkA3XIp4+eFRwuSFQM3DN0fehOlLVfI65aWw=" via-port "1-1" with-interface { 09:00:01 09:00:02 }
6: allow id 0424:ec00 serial "" name "" hash "t3Q3NWJOsmxMC5TAcN4lYL6vjeaHcShDXODSx2q2U08=" parent-hash "E0MWcMEgHt8SbS4HZnYsm7CORelcUwUjAtJKvvAP8yg=" via-port "1-1.1" with-interface ff:00:ff
root@raspberrypi:/root# usbguard list-devices -b
terminate called without an active exception
Aborted
```

dkopecek, 2017-02-16:
Hi, thanks for the report. I've reproduced the bug.

dkopecek, 2017-03-05:
@cgzones Could you please try the latest git version whether you are able to reproduce the problem?

cgzones, 2017-03-07:
when trying to recompile the recent git version on debian sid, i'm getting the following compile errors:
```
/bin/sh ./libtool  --tag=CXX   --mode=compile g++ -DHAVE_CONFIG_H -I. -I./src  -fPIC -I./src -I./src/Library -I./src/Library/IPC -pthread  -Wdate-time -D_FORTIFY_SOURCE=2  -std=c++11 -g -O2 -fdebug-prefix-map=/home/christian/Downloads/usbguard/usbguard-0.6.2+ds1=. -fstack-protector-strong -Wformat -Werror=format-security -fPIC  -fvisibility=hidden  -pedantic -Werror -Wno-unknown-pragmas -Wall -Wextra -Wformat=2 -Wredundant-decls -Wcast-align -Wmissing-declarations -Wmissing-include-dirs -Wmissing-format-attribute -Wswitch-enum -Wswitch-default -Winvalid-pch -Wformat-nonliteral -Wno-missing-field-initializers -Wno-implicit-fallthrough  -Wno-missing-declarations -c -o src/Library/IPC/libusbguard_la-Message.pb.lo `test -f 'src/Library/IPC/Message.pb.cc' || echo './'`src/Library/IPC/Message.pb.cc
libtool: compile:  g++ -DHAVE_CONFIG_H -I. -I./src -fPIC -I./src -I./src/Library -I./src/Library/IPC -pthread -Wdate-time -D_FORTIFY_SOURCE=2 -std=c++11 -g -O2 -fdebug-prefix-map=/home/christian/Downloads/usbguard/usbguard-0.6.2+ds1=. -fstack-protector-strong -Wformat -Werror=format-security -fPIC -fvisibility=hidden -pedantic -Werror -Wno-unknown-pragmas -Wall -Wextra -Wformat=2 -Wredundant-decls -Wcast-align -Wmissing-declarations -Wmissing-include-dirs -Wmissing-format-attribute -Wswitch-enum -Wswitch-default -Winvalid-pch -Wformat-nonliteral -Wno-missing-field-initializers -Wno-implicit-fallthrough -Wno-missing-declarations -c src/Library/IPC/Message.pb.cc  -fPIC -DPIC -o src/Library/IPC/.libs/libusbguard_la-Message.pb.o
src/Library/IPC/Message.pb.cc: In member function ‘virtual google::protobuf::uint8* usbguard::IPC::MessageHeader::InternalSerializeWithCachedSizesToArray(bool, google::protobuf::uint8*) const’:
src/Library/IPC/Message.pb.cc:238:10: error: unused parameter ‘deterministic’ [-Werror=unused-parameter]
     bool deterministic, ::google::protobuf::uint8* target) const {
          ^~~~~~~~~~~~~
/bin/sh ./libtool  --tag=CXX   --mode=compile g++ -DHAVE_CONFIG_H -I. -I./src  -fPIC -I./src -I./src/Library -I./src/Library/IPC -pthread  -Wdate-time -D_FORTIFY_SOURCE=2  -std=c++11 -g -O2 -fdebug-prefix-map=/home/christian/Downloads/usbguard/usbguard-0.6.2+ds1=. -fstack-protector-strong -Wformat -Werror=format-security -fPIC  -fvisibility=hidden  -pedantic -Werror -Wno-unknown-pragmas -Wall -Wextra -Wformat=2 -Wredundant-decls -Wcast-align -Wmissing-declarations -Wmissing-include-dirs -Wmissing-format-attribute -Wswitch-enum -Wswitch-default -Winvalid-pch -Wformat-nonliteral -Wno-missing-field-initializers -Wno-implicit-fallthrough  -Wno-missing-declarations -c -o src/Library/IPC/libusbguard_la-Rule.pb.lo `test -f 'src/Library/IPC/Rule.pb.cc' || echo './'`src/Library/IPC/Rule.pb.cc
libtool: compile:  g++ -DHAVE_CONFIG_H -I. -I./src -fPIC -I./src -I./src/Library -I./src/Library/IPC -pthread -Wdate-time -D_FORTIFY_SOURCE=2 -std=c++11 -g -O2 -fdebug-prefix-map=/home/christian/Downloads/usbguard/usbguard-0.6.2+ds1=. -fstack-protector-strong -Wformat -Werror=format-security -fPIC -fvisibility=hidden -pedantic -Werror -Wno-unknown-pragmas -Wall -Wextra -Wformat=2 -Wredundant-decls -Wcast-align -Wmissing-declarations -Wmissing-include-dirs -Wmissing-format-attribute -Wswitch-enum -Wswitch-default -Winvalid-pch -Wformat-nonliteral -Wno-missing-field-initializers -Wno-implicit-fallthrough -Wno-missing-declarations -c src/Library/IPC/Rule.pb.cc  -fPIC -DPIC -o src/Library/IPC/.libs/libusbguard_la-Rule.pb.o
At global scope:
cc1plus: error: unrecognized command line option ‘-Wno-implicit-fallthrough’ [-Werror]
cc1plus: all warnings being treated as errors
Makefile:1799: recipe for target 'src/Library/IPC/libusbguard_la-Message.pb.lo' failed
make[3]: *** [src/Library/IPC/libusbguard_la-Message.pb.lo] Error 1
make[3]: *** Waiting for unfinished jobs....
src/Library/IPC/Rule.pb.cc: In member function ‘virtual google::protobuf::uint8* usbguard::IPC::Rule::InternalSerializeWithCachedSizesToArray(bool, google::protobuf::uint8*) const’:
src/Library/IPC/Rule.pb.cc:275:10: error: unused parameter ‘deterministic’ [-Werror=unused-parameter]
     bool deterministic, ::google::protobuf::uint8* target) const {
          ^~~~~~~~~~~~~
At global scope:
cc1plus: error: unrecognized command line option ‘-Wno-implicit-fallthrough’ [-Werror]
cc1plus: all warnings being treated as errors
```

dkopecek, 2017-03-08:
@cgzones I made the -Werror flag optional in the latest commit. You should be able to compile it fine now.

dkopecek, 2017-05-10:
Closing as this should be fixed in the latest release.

closed 2017-05-10 by dkopecek